zephyrproject-rtos/zephyr #41488: Stall logging on nrf52840

lmpprk, 2021-12-29:
**Describe the bug**
A clear and concise description of what the bug is.

PR #40735 seems to break the logging on nrf52840dongle.

The console should show logs for a very short while after bootup, but will stall and never show new logs until the device is reboot or reset.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? 
 > nrf52840dongle
 - What have you tried to diagnose or workaround this issue?
 > Revert the commit in PR #40735 then it works again.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the Matter nrf-lighting example for `nrf52840dongle_nrf52840 ` with uart-shell enabled by following [instructions](https://github.com/lmpprk/connectedhomeip/tree/nrf-light/examples/lighting-app/nrfconnect)
2. Load the firmware onto nrf52840dongle
3. Use screen or minicom to observe the logs
4. There will be a few lines of logs but will stall after a short while and never see new logs
5. `kernel reboot cold` or `matter device factoryreset` should reboot the device and you would see a few lines of logs again but stall very quickly again

**Expected behavior**
The log should continue to update as device stay active

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
annoyance, I can't debug

**Logs and console output**

Watch for the timestamp, it should be able to go over a second.

When error observed
```
uart:~$
uart:~$ 7 messages dropped ---
[00:00:00.450,103] <inf> chip: [DL]advertising srp service: 11ED4A406DE81916._matterc._udp
[00:00:00.452,484] <dbg> chip.LogV: [DIS]GetExtendedDiscoveryTimeoutSecs 20
[00:00:00.452,636] <dbg> chip.LogV: [DIS]Scheduling Extended Discovery timeout in secs=20
[00:00:00.555,877] <err> chip: [DL]OnSrpClientNotification: Domain name or RRset is duplicated
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$ matter device factoryreset
Performing factory reset ...
uart:~$ 0.283,020] <inf> chip: [SVR]Copy/paste the below URL in a browser to see the QR Code:
    https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AY3.13OTB00KA0648G00
[00:00:00.287,628] <inf> chip: [SVR]Manual pairing code: [34970112332]
[00:00:00.287,780] <dbg> chip.LogV: [DIS]SetExtendedDiscoveryTimeoutSecs 20
[00:00:00.288,696] <dbg> chip.LogV: [DL]CHIP task running
[00:00:00.291,809] <inf> usb_cdc_acm: Device configured
[00:00:00.296,325] <inf> chip: [DL]CHIPoBLE advertising started
[00:00:00.296,508] <inf> chip: [ZCL]GeneralDiagnosticsDelegate: OnDeviceRebooted
[00:00:00.299,041] <inf> chip: [DL]NFC Tag emulation started
uart:~$
```

When error not observed
```
uart:~$
[00:00:06.055,145] <inf> chip: [DL]BLE connection established (ConnId: 0x00)
[00:00:06.055,267] <inf> chip: [DL]Current number of connections: 1/1
[00:00:06.055,389] <inf> chip: [DL]CHIPoBLE advertising stopped
[00:00:06.055,511] <inf> chip: [DL]NFC Tag emulation stopped
[00:00:06.392,150] <dbg> chip.LogV: [DL]Write request received for CHIPoBLE RX (ConnId 0x00)
[00:00:06.392,272] <inf> chip: [BLE]local and remote recv window sizes = 5
[... 170 lines cut ...]
[00:00:08.557,250] <inf> chip: [ZCL]OpCreds: Fabric 1 was persisted to storage. FabricId 0000000000000001, NodeId 0000000000000059, VendorId 0xFFF1
[00:00:08.557,342] <inf> chip: [ZCL]OpCreds: Call to fabricListChanged
[00:00:08.557,464] <inf> chip: [DIS]Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage
--- 23 messages dropped ---
...
uart:~$ matter onboardingcodes none                                     |                                       |
QRCode:            MT:Y3.13IG000KA0648G00                               +---------------------------------------+
QRCodeUrl:         https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AY3.13IG000KA0648G00
ManualPairingCode: 34970112332
[00:15:00.269,897] <dbg> chip.LogV: [DIS]OnExpiration callback for cleared session
[00:15:00.269,989] <dbg> chip.LogV: [DIS]OnDiscoveryExpiration callback for cleared session
[00:16:06.915,374] <dbg> chip.LogV: [DL]CHIPoBLE advertising set to off
[00:16:06.915,496] <inf> chip: [DL]CHIPoBLE advertising disabled because of timeout expired
[00:16:06.915,924] <inf> chip: [DL]CHIPoBLE advertising stopped
[00:16:06.916,107] <inf> chip: [DL]NFC Tag emulation stopped
[00:24:18.183,563] <err> chip: [DL]OnSrpClientNotification: Domain name or RRset is duplicated
[00:41:45.681,823] <err> chip: [DL]OnSrpClientNotification: Domain name or RRset is duplicated
```

**Environment (please complete the following information):**
 - OS: MacOs
 - Toolchain (e.g Zephyr SDK, ...) NRF Connect SDK 1.8.0, Zephyr 2.7

**Additional context**
Reverting the change in #40735 fixes it for me.

nordic-krch, 2022-01-05:
Can you try to apply commits from https://github.com/zephyrproject-rtos/zephyr/pull/41490/commits and check?

de-nordic, 2022-01-05:
I have also seen the issue after a lot of logging from Bluetooth on device init, no more logs would appear. In my case shell has been disabled and uart has been used for logging only. The platform has also been nrf52840dongle, but I do not think that the platform is a factor here.

nordic-krch, 2022-01-05:
@de-nordic are you able to try that with #41490?

lmpprk, 2022-01-05:
@nordic-krch thanks I tried and it seemed to work correctly now with #41490 applied, I no longer observed the issue.

PR #41490 (2022-01-05): logging: Fix counting of buffered messages
When message is dropped then log_process is called with
bypass flag set and additionally z_log_dropped() is called.
In both functions counter of buffered messages was decremented.
That resulted in counter being decremented twice. It resulted
in logging misbehavior after messages being dropped (delayed
processing). Fixing it by decrementing the counter in log_process
only when bypass flag is not set.

Additionally, waking up algorithm is improved by using value returned by `atomic_inc`
instead of reading variable directly.

Fixes #41240.
Fixes #41488.
PR #42541 (2022-02-08): [backport2.7] logging: Fix tracking of active messages
Backporting #40735 and #41969 (added later as first fix was not complete).

Fixes #41488 for 2.7.
closed 2022-01-05 by carlescufi