zephyrproject-rtos/zephyr #41489: Make POSIX subsystem available while using an external libc

yashi, 2021-12-29:
**Is your enhancement proposal related to a problem? Please describe.**
Some functions under `lib/posix/` need to access kernel services.  These functions must be built on top of Zephyr kernel API.  Thus, we should provide them even with external libc.

**Describe the solution you'd like**
- Make POSIX subsystem accessible while an external libc is selected.
- Make sure all functions can be configurable so that there won't be any name collision with external libc.

**Additional context**
I'm trying to make picolibc as a Zephyr module.  I'd like to use the POSIX subsystem while using picolibc.

yashi, 2021-12-29:
Is something like this acceptable?

```diff
diff --git a/include/sys/libc-hooks.h b/include/sys/libc-hooks.h
index afa3554ed87..90ca4ffbe68 100644
--- a/include/sys/libc-hooks.h
+++ b/include/sys/libc-hooks.h
@@ -16,7 +16,7 @@
  * that need to call into the kernel as system calls
  */
 
-#if defined(CONFIG_NEWLIB_LIBC) || defined(CONFIG_ARCMWDT_LIBC)
+#if defined(CONFIG_NEWLIB_LIBC) || defined(CONFIG_ARCMWDT_LIBC) || defined(CONFIG_PICOLIBC)
 
 /* syscall generation ignores preprocessor, ensure this is defined to ensure
  * we don't have compile errors
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index ca8c4b9440a..8fb12a046d1 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -2,8 +2,8 @@
 
 if(NOT CONFIG_EXTERNAL_LIBC)
 add_subdirectory(libc)
-add_subdirectory(posix)
 endif()
+add_subdirectory(posix)
 add_subdirectory(gui)
 add_subdirectory(os)
 add_subdirectory(util)
```

cfriedt, 2022-11-22:
This should be fixed by LTSv3 (see [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211))

cfriedt, 2023-06-24:
@stephanosio - this might Just Work™️ already. 
@yashi - can you provide a reproduction recipe? E.g. a single `west` or `twister` command that can be used to demonstrate the problem? It sounds suspiciously like a bug, but I would need a bit more information to make that call.

cfriedt, 2023-06-24:
@keith-packard - do you by any chance have any commentary on this issue?

Personally, I think it's touching on a couple of related areas that may not be super obvious, but one is that POSIX should work on systems with an external C library (which implies external C library headers).

IIRC, this is the main reason we opted to copy POSIX headers in to Zephyr.

keith-packard, 2023-06-25:
It's gonna be tricky if the external C library also offers POSIX functions -- I suspect there will be cases needing changes in the C library to make things work correctly.

cfriedt, 2024-01-30:
This should be fixed as of 6342aa3cc0

closed 2024-01-30 by cfriedt