zephyrproject-rtos/zephyr #41509: OpenThread's timer processing enters infinite loop in 49th day of system uptime

hubertmis, 2021-12-30:
**Describe the bug**
My devices based on nRF52840 are running with Zephyr RTOS with OpenThread stack. They all hang if uptime reaches 49 days. Today I've attached a debugger and found the root cause of the bug.

The device hangs, because main workqueue enters infinite loop. It is caused by OpenThread's alarm processing using the workqueue. The alarm processing functions schedule next timer event to be processed immediately instead of waiting until the timer elapses. Immediate event processing adds a job to the main workqueue causing processing of the main workqueue. The workqueue job processes OpenThread's alarm which schedules timer event immediately (instead of waiting) adding a job to the workqueue, and the cycle continues.

It looks that the problem (processing timer event immediately instead of waiting) is caused by incorrectly handled integer overflow in https://github.com/zephyrproject-rtos/zephyr/blob/339a6bdafbe558e3ad0ea57336e2ead62e2237ac/subsys/net/lib/openthread/platform/alarm.c#L79 . OpenThread's API uses 32-bit millisecond time base (which overflows in 49th day of uptime), while Zephyr uses 64-bit millisecond time base, which does not overflow in reasonable uptime. There is naive approach implemented by casting 32-bit add result to 64-bit variable, but it loses the msb word. It produces invalid `delta` value after timer overflow, what causes the algorithm misbehavior.

**To Reproduce**
Steps to reproduce the behavior:
1. Build any sample/application with OpenThread stack
2. Wait 50 days
3. See that the device hangs in the main workqueue processing

**Expected behavior**
It is expected that the device correctly handles timer overflows and can operate within whole 64-bit millisecond timebase.

**Impact**
I need to reset my devices every 50 days, what's annoying.
I believe it might be a showstopper for commercial products.

**Additional context**
It looks similar issue is fixed in the upstream bare-metal OpenThread's alarm implementation where missing most significant word is restored from 64-bit system timer:
https://github.com/openthread/ot-nrf528xx/blob/b816fe1a1a534fea5f97500200d515cf6338acb9/src/src/alarm.c#L371

hubertmis, 2021-12-30:
FYI @edmont @lududa

aurel32, 2021-12-30:
Which version of Zephyr are you running? This is probably the same bug as #39704, which is fixed in master and in Zephyr 2.7.1.

hubertmis, 2021-12-30:
Good point. I don't know why I've been looking at 339a6bd commit instead of main. It looks the issue is fixed on main. I'll upgrade my devices and reopen if the problem reoccurs

closed 2021-12-30 by hubertmis