zephyrproject-rtos/zephyr #41523: drivers: i2c: i2c_mcux: unused variables cause daily build failures

cfriedt, 2022-01-01:
**Describe the bug**
Twister fails during the daily build due to unused variables (warnings promoted to errors).

**To Reproduce**
```
twister -i -p ip_k66f -s samples/subsys/display/grove_display/sample.display.grove_display
```

**Expected behavior**
Daily build should not fail.

**Impact**
Annoyance, because the build succeeds with
```
west build -p auto -b ip_k66f samples/subsys/display/grove_display
```

**Logs and console output**
```
/__w/zephyr/zephyr/drivers/i2c/i2c_mcux.c:216:36: error: 'i2c_mcux_driver_api' defined but not used [-Werror=unused-const-variable=]
  216 | static const struct i2c_driver_api i2c_mcux_driver_api = {
      |                                    ^~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/i2c/i2c_mcux.c:186:12: error: 'i2c_mcux_init' defined but not used [-Werror=unused-function]
  186 | static int i2c_mcux_init(const struct device *dev)
      |            ^~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/i2c/i2c_mcux.c:178:13: error: 'i2c_mcux_isr' defined but not used [-Werror=unused-function]
  178 | static void i2c_mcux_isr(const struct device *dev)
      |             ^~~~~~~~~~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: CI
 - Toolchain: Zephyr SDK 0.13.2
 - Commit SHA: 7f5eb921dc8bb1b7014886986bbffdf111b68b15

**Additional context**
Daily Build failure:
https://github.com/zephyrproject-rtos/zephyr/runs/4676613761?check_suite_focus=true

dleach02, 2022-01-05:
@cfriedt, FYI I get the same warnings with 'west'

cfriedt, 2022-01-05:
> @cfriedt, FYI I get the same warnings with 'west'

@dleach02 - yes, but they aren't promoted to errors. We should try to eliminate the warnings too

dleach02, 2022-01-05:
well that is because twister forces warnings as errors... but the same warnings.

but what is needed is a board specific overlay on how this display is connected to the board. We have never connected this display to the board in question. I would like to have something in the "samples.yaml" to exclude targets that don't have an overlay or targets that haven't enabled the i2c. Right now it builds for all boards that say they support i2c, which this board does, but the device node is not enabled by default. So is there a way to set the yaml to know this?

dleach02, 2022-01-25:
this display (grove) and test combination is problematic. The test yaml is only filtering on I2C because the display device object is create programmatically in the grove display driver and not via the device tree so there is nothing to filter on in the test case to check if the display is part of your hardware configuration. 

You could filter on an instance of I2C_0 label since that is what they give as the grove device name but that is a problem since the underlying device tree filtering doesn't have support routines to select device objects with a property "label".

What should probably be done here is to modify the grove display to be defined via the devicetree and then provide an overlay for the target board(s) that include and enable this. I suspect that the display properties could be fairly simple to accomplish this task but I don't have a board to verify/test this.

dleach02, 2022-01-26:
@cfriedt check out the PR... I tested to ensure that it built the platforms that support an enabled I2C_0 devicetree node and skipped those that do not have one (i.e. node.status is "okay")

PR #42234 (2022-01-28): samples: grove_display: Exclude ip_k66f from sample build
The ip_k66f doesn't have I2C_0 status set to OKAY so the build of
this sample generates warnings in the nightly build. Root problem
is that the groove display is not ported to the device tree so
there isn't a standard way of filtering on the display being
part of the platform.

fixes #41523
commit 96599830 (2022-01-28): samples: grove_display: Exclude ip_k66f from sample build

The ip_k66f doesn't have I2C_0 status set to OKAY so the build of
this sample generates warnings in the nightly build. Root problem
is that the groove display is not ported to the device tree so
there isn't a standard way of filtering on the display being
part of the platform.

fixes #41523
closed 2022-01-28 by dleach02