zephyrproject-rtos/zephyr #41557: ESP32 Uart 2-bit Stop Register Setting

huarunlin, 2022-01-04:
While working on MODBUS implementation with ESP32 I discovered an issue with the ESP32 UART driver. Refer to ESP-IDF code and chip manual, If there are two stop bits, set UART_STOP_BIT_NUM in UART_CONF0_REG to 1 and UART_DL1_EN in UART_RS485_CONF_REG to 1.

esp-idf v4.3.1 components\hal\esp32\include\hal\uart_ll.h

```
FORCE_INLINE_ATTR void uart_ll_set_stop_bits(uart_dev_t *hw, uart_stop_bits_t stop_bit)
{
    //workaround for hardware issue, when UART stop bit set as 2-bit mode.
    if(stop_bit == UART_STOP_BITS_2)  {
        hw->rs485_conf.dl1_en = 1;
        hw->conf0.stop_bit_num = 0x1;
    } else {
        hw->rs485_conf.dl1_en = 0;
        hw->conf0.stop_bit_num = stop_bit;
    }
}
```

Therefore, the uart_ESP32_configure function in the ESP32 UART driver needs to be changed to

```
static int uart_esp32_configure(const struct device *dev,
				const struct uart_config *cfg)
{
	uint32_t rs485_conf = DEV_BASE(dev)->rs485_conf;
	uint32_t conf0 = UART_TICK_REF_ALWAYS_ON;
	uint32_t conf1 = (UART_RX_FIFO_THRESH << UART_RXFIFO_FULL_THRHD_S)
		      | (UART_TX_FIFO_THRESH << UART_TXFIFO_EMPTY_THRHD_S);

	uart_esp32_configure_pins(dev);
	clock_control_on(DEV_CFG(dev)->clock_dev, DEV_CFG(dev)->clock_subsys);

	/*
	 * Reset RX Buffer by reading all received bytes
	 * Hardware Reset functionality can't be used with UART 1/2
	 */
	while (UART_RXFIFO_COUNT(DEV_BASE(dev)->status) != 0) {
		(void) DEV_BASE(dev)->fifo;
	}

	switch (cfg->parity) {
	case UART_CFG_PARITY_NONE:
		conf0 &= ~(UART_PARITY_EN);
		conf0 &= ~(UART_PARITY);
		break;
	case UART_CFG_PARITY_EVEN:
		conf0 &= ~(UART_PARITY);
		break;
	case UART_CFG_PARITY_ODD:
		conf0 |= UART_PARITY;
		break;
	default:
		return -ENOTSUP;
	}

	switch (cfg->stop_bits) {
	case UART_CFG_STOP_BITS_1:
	case UART_CFG_STOP_BITS_1_5:
		rs485_conf &= ~UART_DL1_EN;
		conf0 |= cfg->stop_bits << UART_STOP_BIT_NUM_S;
		break;
	case UART_CFG_STOP_BITS_2:
		rs485_conf |= UART_DL1_EN;
		conf0 |= 1 << UART_STOP_BIT_NUM_S;
		break;
	default:
		return -ENOTSUP;
	}

	if (cfg->data_bits <= UART_CFG_DATA_BITS_8) {
		conf0 |= cfg->data_bits << UART_BIT_NUM_S;
	} else {
		return -ENOTSUP;
	}

	switch (cfg->flow_ctrl) {
	case UART_CFG_FLOW_CTRL_NONE:
		conf0 &= ~(UART_TX_FLOW_EN);
		conf1 &= ~(UART_RX_FLOW_EN);
		break;
	case UART_CFG_FLOW_CTRL_RTS_CTS:
		conf0 |= UART_TX_FLOW_EN;
		conf1 |= UART_RX_FLOW_EN;
		break;
	default:
		return -ENOTSUP;
	}

	if (uart_esp32_set_baudrate(dev, cfg->baudrate)) {
		DEV_DATA(dev)->uart_config.baudrate = cfg->baudrate;
	} else {
		return -ENOTSUP;
	}

	DEV_BASE(dev)->rs485_conf = rs485_conf;
	DEV_BASE(dev)->conf0 = conf0;
	DEV_BASE(dev)->conf1 = conf1;

	return 0;
}
```

huarunlin, 2022-01-05:
@sylvioalves Release v2.7.1
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.7.1/drivers/serial/uart_esp32.c#L286

huarunlin, 2022-01-05:
> @huarunlin, this should be working properly on main branch. Which revision have you been working with? https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_esp32.c#L295

@sylvioalves Release v2.7.1
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.7.1/drivers/serial/uart_esp32.c#L286

sylvioalves, 2022-01-05:
@huarunlin So, the uart driver has been refactored to use HAL layer. However, this got merged after 2.7 release.  
1) You can use latest main which have this in place.  
2) As you already proposed the solution, you could open a PR to 2.7 release (backport). @cfriedt Can this fix be backported? I am not sure as this is not a fix on main branch.

sylvioalves, 2022-01-06:
Can we close this?

cfriedt, 2022-01-06:
@sylvioalves , @huarunlin - if there is a bug in this driver for v2.7.1, please create a PR to fix and link against this issue. Otherwise, I would suggest using main. We can only backport bugfixes to `v2.7-branch`.

closed 2022-01-07 by huarunlin