zephyrproject-rtos/zephyr #41581: STM32 subghzspi fails pinctrl setup

abxy, 2022-01-05:
**Describe the bug**
The subghzspi driver fails during initialization when setting up pinctrl.

It fails because subghzspi doesn't have any pinctrl states so `pinctrl_apply_state(..., PINCTRL_STATE_DEFAULT)` returns `-ENOENT` to indicate that the "default" state was not found, which is treated as an error by the driver. Modifying the code so that `-ENOENT` is not treated as an error or so that pinctrl setup is not done if `pcfg->state_cnt == 0` solves the problem.

I attempted a workaround by adding some "no-op" pinctrl in the subghzspi node, but this is not possible because `pinctrl-0` and `pinctrl-names` are blacklisted by its devicetree binding.

**To Reproduce**
Build and run the `subsys/lorawan/class_a` sample for `nucleo_stm32wl55jc` on a recent main.

```
*** Booting Zephyr OS build v2.7.99-2798-g66fef3123625  ***
[00:00:00.000,000] <err> spi_ll_stm32: SPI pinctrl setup failed (-2)
[00:00:00.000,000] <err> sx126x: Cannot get pointer to SUBGHZSPI device
[00:00:00.035,000] <err> os: ***** BUS FAULT *****
[00:00:00.035,000] <err> os:   Instruction bus error
[00:00:00.035,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x200005a4  r2/a3:  0x20001f80
[00:00:00.035,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x08003a4f
[00:00:00.035,000] <err> os:  xpsr:  0x61000000
[00:00:00.035,000] <err> os: Faulting instruction address (r15/pc): 0x09f7f6b4
[00:00:00.035,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.035,000] <err> os: Current thread: 0x20000348 (unknown)
[00:00:00.091,000] <err> os: Halting system
```

erwango, 2022-01-05:
@abxy Thanks for reporting this.

I think that the following looks like the better approach
> that pinctrl setup is not done `if pcfg->state_cnt == 0`

Though, it should be applied only if instance is a st,stm32-spi-subghz, 
so we'd need to add something like `if (cfg->use_subghzspi_nss)` to verify it.

abxy, 2022-01-05:
Thanks for the input. 

Can `cfg->use_subghzspi_nss` really be used reliably for this? A subghzspi device that doesn't use that option would still fail the same way.

I wonder if it would make more sense to test if the device has the subghzi compat during instantiation (using `DT_NODE_HAS_COMPAT([...], st_stm32_spi_subghz)`)  and not define a `struct pinctrl_dev_config` for it. That would leave `pcfg == NULL` which could be tested by the code during init.

erwango, 2022-01-05:
> Can cfg->use_subghzspi_nss really be used reliably for this?

Since it is defined in `dts/arm/st/wl/stm32wl.dtsi`, I'm under the impression a subghzspi device has to use this option.
Maybe @fabiobaltieri can confirm ?

> I wonder if it would make more sense to test if the device has the subghzi compat during instantiation

This would apply to all instances, but this process should not be applied to other, non subghz, SPI instances.

abxy, 2022-01-05:
I think I misunderstood the implication of a boolean devicetree property with `required: true`.  It seems like there is no way to set it to false. So `cfg->use_subghzspi_nss` should be a reliable test. I'll make a PR using this approach.

> This would apply to all instances, but this process should not be applied to other, non subghz, SPI instances.

It can be done per instance using the `COND_CODE_0` macro. I did a test after proposing the idea. But I prefer to not use such macrobatics if it can be avoided.

fabiobaltieri, 2022-01-05:
> > Can cfg->use_subghzspi_nss really be used reliably for this?
> 
> Since it is defined in `dts/arm/st/wl/stm32wl.dtsi`, I'm under the impression a subghzspi device has to use this option. Maybe @fabiobaltieri can confirm ?

Yeah that's the case with the current hardware (the way that the WL chip is built). I suppose one could imagine a device with an internal SPI with the chip select going through a GPIO (wired internally), but then it would just look like a normal SPI, would have a gpio phandle and everything else should work normally. This sounds like a reasonable approach to me.

> 
> > I wonder if it would make more sense to test if the device has the subghzi compat during instantiation
> 
> This would apply to all instances, but this process should not be applied to other, non subghz, SPI instances.

closed 2022-01-11 by carlescufi

tcpipchip, 2022-11-07:
hi
Do you know if this bug was solved ?
Can you share us the code ?
I got here a STM32WL55 to test!
Thanks!