zephyrproject-rtos/zephyr #41597: Unable to build mcuboot for BL654_DVK

realszopen, 2022-01-05:
**Describe the bug**
A clear and concise description of what the bug is.

I am unable to build mcuboot bootloader for bl654_dvk. I can setup the build folder, cmake configure runs fine, but when trying to build with Ninja I get:

FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map /home/szopen/mcuboot/boot/zephyr/build/zephyr/zephyr_prebuilt.map 
: && ccache /home/szopen/gnuarmemb/bin/arm-none-eabi-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/szopen/mcuboot/boot/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  modules/nrf/drivers/hw_cc310/lib..__nrf__drivers__hw_cc310.a  modules/mcuboot/boot/bootutil/zephyr/libmcuboot_util.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/mbedtls/libmodules__mbedtls.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m/nofp"  -L/home/szopen/mcuboot/boot/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  /home/szopen/ncs/nrfxlib/crypto/nrf_cc310_platform/lib/cortex-m4/soft-float/no-interrupts/libnrf_cc310_platform_0.9.11.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/szopen/mcuboot/boot/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_mutex_lock':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:724: undefined reference to `z_impl_k_mutex_lock'
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_sem_take':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:826: undefined reference to `z_impl_k_sem_take'
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_mutex_unlock':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:738: undefined reference to `z_impl_k_mutex_unlock'
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_sem_init':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:810: undefined reference to `z_impl_k_sem_init'
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_mutex_init':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:708: undefined reference to `z_impl_k_mutex_init'
/home/szopen/gnuarmemb/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_sem_give':
/home/szopen/mcuboot/boot/zephyr/build/zephyr/include/generated/syscalls/kernel.h:841: undefined reference to `z_impl_k_sem_give'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

**To Reproduce**
Steps to reproduce the behavior:
1. cd
2. git clone https://github.com/mcu-tools/mcuboot
3. cd mcuboot/boot/zephyr
4. mkdir build
5. cd build
6. cmake -GNinja -DBOARD=bl654_dvk ..
7. ninja
8. See the above error

**Expected behavior**
Expected behaviour is the build to fail. Wanted behaviour is the build to succeed.

**Impact**
Unable to build the bootloader is a showstopper or major annoyance and hindrance to the project.

**Logs and console output**
Above.

**Environment (please complete the following information):**
Linux - OpenSUSE Tumbleweed

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

lairdjm, 2022-01-12:
@anangl how is best to handle this? Our board enables the 32KHz RC source by default so it works with the radio samples, but by doing that it's pulling the temperature compensation/check code into mcuboot where it isn't needed, and it won't select the multithreading option by default for mcuboot (this fix will need applying to multiple of our boards with don't have 32KHz crystals fitted)

lairdjm, 2022-01-21:
@carlescufi can anyone at nordic answer the above query on this bug?

realszopen, 2022-01-28:
@carlescufi Can anyone please answer the above?

jfischer-no, 2022-01-31:
What about `cmake -DBOARD=bl654_dvk -DCONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n ../` ?

lairdjm, 2022-02-01:
@jfischer-no this requires a specific build command be used when building mcuboot and will not work when using nRF connect SDK where mcuboot is an image, it would be preferable if this could be fixed without supplying config parameters via the command line.

carlescufi, 2022-03-10:
@nordic-krch @nvlsianpu 

I see 4 ways out of this, from what I gather:

- Make the code for the temp sensor compatible with no-multithreading (IMHO overkill)
- Select `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n` in `mcuboot/tree/main/boot/zephyr/boards/bl654_dvk.conf`
- Select `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n` in `mcuboot/tree/main/boot/zephyr/Kconfig`
- Select `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n` somewhere in the main zephyr tree if `CONFIG_MULTITHREADING=n` (my preference)

PR #43667 (2022-03-11): drivers: clock_control: nrf: Fix missing dependency to mt
The temperature sensor used in the clock_control driver requires
multithreading, but this is not compatible with mcuboot builds with
multithreading disabled.

Fixes #41597.
commit af9a6a69 (2022-03-11): drivers: clock_control: nrf: Fix missing dependency to mt

The temperature sensor used in the clock_control driver requires
multithreading, but this is not compatible with mcuboot builds with
multithreading disabled.

Fixes #41597.
closed 2022-03-11 by MaureenHelm