zephyrproject-rtos/zephyr #41613: Process: Review and update Milestone Definitions

carlescufi, 2022-01-05:
The Milestone Definitions are not being respected correctly in our current release process.
Adapt and rework them so that they make sense to our releases:

https://github.com/zephyrproject-rtos/zephyr/wiki/Program-Management#milestone-definitions

mbolivar-nordic, 2022-01-26:
Minutes:

- @carlescufi :  milestones as defined in the wiki do not reflect what we do now. Should we refine this to reflect what we do now without having to reorganize ourselves. E.g. 'planning kickoff' could be replaced with 'talking about next release at first TSC meeting following a release'
- @dleach02 : we are time-based, so having a milestone to do planning based on P0/P1 etc doesn't happen. Each subsystem or technology silo goes on its own cadence, and I think that's okay.
- @carlescufi : it's still useful to have an answer to "what are you doing for v3.1"
- @nashif : things like release managers should be agreed on a long time before the release, agreeing on dates, etc. That's what the planning is. Probably some of the language needs changing.
- @dleach02 : yes, and the language is around features that go into releases. We are time based, so we just move features around based on whether they're ready or not. We should separate features from planning because I think it doesn't matter
- @carlescufi : I think we should have a project for major features we want, but I'm not sure that having this in the release process is a good idea
- @dleach02 : If you think about things like API changes, the complication is that we're looking at this as a per-release process flow, but there are all these tech silos that look out ahead in order to fit our time-based release schedule
- @mbolivar-nordic : I would say that having a release plan project on github with features assigned to versions is misleading, since we only merge when things are ready, and release go out on schedules
- @dleach02 : I agree that LTS needs more discussion but is separate
- @carlescufi : I'll take an action to reword this and make a proposal

carlescufi, 2022-02-14:
```mermaid
  graph TD;
      A-->B;
      A-->C;
      B-->D;
      C-->D;
```

mbolivar-nordic, 2022-02-23:
Process WG:

- @nashif: I went through https://docs.zephyrproject.org/latest/development_process/release_process.html, which was a foundational document within the project. We usually refer to the wiki but this is part of our documentation. I recommend using this instead of the wiki in the future.
- @mbolivar-nordic : let's take that as a decision -- no objections, so we're going to remove the release process notes from the wiki and just use the docs from now on.
- General consensus that @nashif's document is going in the right direction; will continue offline and continue discussion next week

nashif, 2022-03-01:
new release timeline: https://docs.google.com/document/d/1JxFd2FIR1hVbwA8q1N8u4JV0719aL-omZgghlCyYSAU/edit?usp=sharing

mbolivar-nordic, 2022-03-02:
Process WG: continue discussion in the Google Doc. Overall trending in the right direction. Open questions remain regarding assigning milestones to issues and PRs at a project level.

nashif, 2022-03-03:
updated table:

[image]

mbolivar-nordic, 2022-03-09:
Process WG:

- agreement on the above proposal for milestone usage
- @nashif : should maybe add a bot that removes milestones applied by anyone not in the right group
- @nashif : program management wiki page has been trimmed down; AI to update document based on above discussion
- @nashif : people make requests for setting the milestone by setting the label; it gets triaged at that meeting at T-7W. No more issues can get that milestone within the current cycle
- Users with triage rights can continue to set the milestone, but anything that isn't a blocker bug for R0 will get rejected after T-7W

nashif, 2022-03-30:
#44326

mbolivar-nordic, 2022-03-30:
Process WG:

- @nashif: see above PR
- All to review the PR and discuss at TSC next week

mbolivar-nordic, 2022-04-06:
Process WG:

- @nashif: PR is ready for review

PR #44326 (2022-04-19): doc: release milestone definition overhaul
x- doc: releases: replace merge window terminology
- doc: releases: remove moderation levels
- doc: release process: update milestone definition

Fixes: #41613
closed 2022-04-19 by nashif