zephyrproject-rtos/zephyr #41616: Define process for managing access to essential resources of the project

nashif, 2022-01-05:
Define a process for managing access with elevated permission levels to essential resources of the project (Github, CI, communication platforms, websites, etc.):

- Define requirements for having access
- Define requirements for maintaining access
- Define when access should be revoked
- Define backup plans

mbolivar-nordic, 2022-01-19:
Process WG minutes:

- @nashif: urgent and was discussed at TSC. E.g. @galak was listed as a GH owner and is currently on sabbatical. We need a checklist for what happens when someone leaves.
- @dleach02 : perhaps a periodic meeting to review ACLs in addition to checklists, along with documentation of what the infrastructure is
- @nashif : backup plans: if something happens to me, I may be a single point of failure in some areas now that Kumar is gone.

mbolivar-nordic, 2022-02-09:
Process WG:

- @mbolivar-nordic : agree with the areas requiring access; what about making escrow for LF explicit for each?
- @nashif : we have that for github, AWS, and Discord. Agree we need to do that across the board
- @dleach02 : let's document that as a general guideline where possible. An example where it's problematic right now is the license for maxwell pro is retained by NXP, but the PR was paid for by the LF
- @nashif : will work with Brett to list initial resources and who has access. Also working on hiring someone to own the infrastructure; hopefully they can take over.

mbolivar-nordic, 2022-02-23:
Process WG:

- put on hold pending hiring dev ops engineer at LF

mbolivar-nordic, 2022-03-30:
Process WG: reactivated the issue, reassigned given @stephanosio is hired now

stephanosio, 2022-03-30:
I will join the Process meeting next week to discuss this.

mbolivar-nordic, 2022-04-04:
@stephanosio thanks! Before then, can you and @nashif try to find some time to put the framework in place, so we have something to review and discuss at the WG?

stephanosio, 2022-04-06:
@mbolivar-nordic sure, I will let you know when we have something substantial enough to be discussed at the WG.

mbolivar-nordic, 2022-04-06:
Process WG:

- @stephanosio : is this public documentation or internal?
- @nashif : internal for the most part; list of people who have access to individual areas doesn't have to be public. General guidelines can be shared with the TSC.
- @stephanosio : do we need a formalization of what exists now, or something new?
- @mbolivar-nordic : IMO we need a new formal process especially for revoking access
- @stephanosio : I'll come up with a plan with @nashif and present here
- @nashif: let's start with the most critical services first, e.g. checking AWS users are active and doing work and revoking access for inactive users

mbolivar-nordic, 2022-04-13:
Process WG:

- @stephanosio :  we had a brainstorm in the infrastructure channel on Discord. Idea is to have a single source of truth for everything -- one YAML file with all resources and a python script that goes through github configurations, permissions, AWS resources, and checks configurations in YAML against actual configurations in live systems. Not finalized; still thinking of the best way to do this. Any suggestions or objections welcome.
- @nashif : need to iron out policy before getting to implementation
- @nashif : we have owner and can report in TSC under @stephanosio 's ownership
- Moved to on hold for process WG purposes

mbolivar-nordic, 2022-05-04:
@stephanosio can you please make sure that edit access to groups.io events is also on the list of resources, so meeting chairs can cancel events as needed?

stephanosio, 2022-05-04:
> @stephanosio can you please make sure that edit access to groups.io events is also on the list of resources, so meeting chairs can cancel events as needed?

@mbolivar-nordic unfortunately, groups.io is one of the few services that I do not have admin access to (it is managed by the Linux Foundation). I will ask @bprestonlf to take a look.

keith-zephyr, 2024-06-26:
@nashif to follow up with @stephanosio for next steps.  Removing from Process WG agenda.