zephyrproject-rtos/zephyr #41624: ESP32 Uart uart_esp32_irq_tx_ready

huarunlin, 2022-01-06:
The modbus serial uart_cb_handler callback function handles all uart interrupt events. In recv state (rx irq enable, tx irq disable), cb_handler_rx obtains the data, and then judges uart_irq_tx_ready (esp32 uart is realized as judgment tx fifo len> 0). RS485 feature, at this time fifo len is 0, the data just received will be It was sent immediately, causing it to fail to receive normally.

https://github.com/zephyrproject-rtos/zephyr/blob/c9a3dae6519fc788d33233fc1c4d1364ba129034/subsys/modbus/modbus_serial.c#L422

https://github.com/zephyrproject-rtos/zephyr/blob/c9a3dae6519fc788d33233fc1c4d1364ba129034/drivers/serial/uart_esp32.c#L394

The uart_esp32_irq_tx_ready in the esp32 uart driver should be modified to:

```
static int uart_esp32_irq_tx_ready(const struct device *dev)
{
    return (uart_hal_get_txfifo_len(&DEV_CFG(dev)->hal) > 0 && uart_hal_get_intr_ena_status(&DEV_CFG(dev)->hal) & UART_TXFIFO_EMPTY_INT_ENA);
}
```

huarunlin, 2022-01-11:
> @huarunlin, correct. I think that `UART_TXFIFO_EMPTY_INT_ENA` should be replaced to `UART_INTR_TXFIFO_EMPTY`. I am submitting this.

@sylvioalves I tried to profile UART_TXFIFO_EMPTY_INT_RAW before and found that UART_TXFIFO_EMPTY_INT_RAW is still flagged when UART_TXFIFO_EMPTY_INT_ENA is disabled. Just now I re-read the manual and esp-idf, and found that the correct way should be to evaluate the UART_TXFIFO_EMPTY_INT_ST flag. If you use uart_hal_get_intsts_mask() to evaluate UART_INTR_TXFIFO_EMPTY, that should be correct.

huarunlin, 2022-01-11:
> @huarunlin, correct. I think that `UART_TXFIFO_EMPTY_INT_ENA` should be replaced to `UART_INTR_TXFIFO_EMPTY`. I am submitting this.

@sylvioalves I tried to profile UART_TXFIFO_EMPTY_INT_RAW before and found that `UART_TXFIFO_EMPTY_INT_RAW `is still flagged when `UART_TXFIFO_EMPTY_INT_ENA `is disabled. Just now I re-read the manual and esp-idf, and found that the correct way should be to evaluate the `UART_TXFIFO_EMPTY_INT_ST `flag. If you use `uart_hal_get_intsts_mask()` to evaluate `UART_INTR_TXFIFO_EMPTY`, that should be correct.

sylvioalves, 2022-01-18:
@dkalowsk can you check this? #41892

PR #41892 (2022-01-18): drivers: uart: fix esp32 TX checks
This fixes both TX ready and TX completed UART
calls to meet valid condition.

Closes #41526
Closes #41624
commit 40ad264a (2022-01-18): drivers: uart: fix esp32 TX checks

This fixes both TX ready and completed uart
calls to meet valid condition.

Closes #41526
Closes #41624
closed 2022-01-18 by nashif