zephyrproject-rtos/zephyr #41628: Move LVGL glue code to zephyr/modules/

brgl, 2022-01-06:
The in-tree glue code for LVGL should live in zephyr/modules/ and not in lib/gui/lvgl as it requires an external module repo.

This is a follow-up task to https://github.com/zephyrproject-rtos/zephyr/pull/41336

tejlmand, 2022-01-06:
@brgl as no one has been assigned to this task could you please provide a detailed description of:
- What files are considered glue files and must be moved to [zephyr/modules](https://github.com/zephyrproject-rtos/zephyr/tree/main/modules)
- Are there any settings in those files that must be moved to another Kconfig file and kept in the `lib/gui` tree, as they are not considered glue code
- Other relevant pieces of information so the person assigned know what to do.

brgl, 2022-01-06:
> @brgl as no one has been assigned to this task could you please provide a detailed description of:
>
> * What files are considered glue files and must be moved to [...]

I would have assigned myself if I could but I don't have the permissions. Will do that tomorrow, sure. Please assign it to me if you can.

tejlmand, 2022-01-06:
> I would have assigned myself 

Done, and in that case no need for detailed info.
Was only concerned that if this task would stay idle for some time, then someone would be unsure exactly what was required.

brgl, 2022-03-22:
This has been implemented in https://github.com/zephyrproject-rtos/zephyr/pull/41336