zephyrproject-rtos/zephyr #41637: Modbus Gateway: Transaction ID Error!

maweicoltsmart, 2022-01-07:
samples/subsys/modbus/tcp_gateway/

fix the bug:
```
diff --git a/subsys/modbus/modbus_raw.c b/subsys/modbus/modbus_raw.c
index e4e330f5ad..e283dc7f8f 100644
--- a/subsys/modbus/modbus_raw.c
+++ b/subsys/modbus/modbus_raw.c
@@ -147,7 +147,7 @@ int modbus_raw_backend_txn(const int iface, struct modbus_adu *adu)
        err = modbus_tx_wait_rx_adu(ctx);
 
        if (err == 0) {
-               memcpy(adu, &ctx->rx_adu, sizeof(struct modbus_adu));
+               memcpy((uint8_t *)adu + 4, (uint8_t *)&ctx->rx_adu + 4, sizeof(struct modbus_adu) - 4);
        } else {
                modbus_set_exception(adu, MODBUS_EXC_GW_TARGET_FAILED_TO_RESP);
        }

```

jukkar, 2022-01-07:
Please send a PR for the fix so that it can be reviewed and merged properly.

jfischer-no, 2022-01-07:
> Please send a PR for the fix so that it can be reviewed and merged properly.

It does not look like correct fix for me. And for what bug?

jukkar, 2022-01-07:
> > Please send a PR for the fix so that it can be reviewed and merged properly.
> 
> It does not look like correct fix to me. And for what bug?

I do not know, that is why I suggested that a PR is created so that the "fix" can be reviewed.

PR #41671 (2022-01-10): modbus: keep transaction and protocol IDs in gateway mode
Serial line interface does not use transaction and protocol
IDs. Continue to use memcpy() for the whole frame as it is safer
and more efficient but temporarily store transaction and protocol
IDs, and write it back if the transfer was successful.

Fixes: #41637
closed 2022-01-10 by MaureenHelm