zephyrproject-rtos/zephyr #41650: STM32H7 SPI123 incorrect clock source used for prescaler calculation

heinwessels, 2022-01-07:
## Describe the bug
In the STM32H7 series the assumed clock source is incorrect when using SPI1, 2 or 3. This incorrect clock source is then used to calculate the clock prescaler (`SPI_CFG:MBR`), which can then potentially be calculated incorrectly. More specifically, it assumes the SCK clock source is, for example `APB2`, while it's by actually by default `PLL1Q`. This resulted that our custom STM32H743 implementation is outputting an incorrect SPI clock frequency.

This is because for the H7 series the SPI1, 2, and 3 has a _seperate_ clock domain that governs the SCK speed. See the following snippet from RM0433 Figure 609. 

[image]

Here `spi_ker_ck` is the `SCLK` source, not `PCLK` (f.i. `APB2`). The value of the `spi_ker_ck` is set by `RCC_D2CCIP1R : SPI123SEL`, which is `pll1_q_ck` by default.

## To Reproduce
We can slightly adjust the `nucleo_h743zi` DTS and abuse the `led_ws2812` app to easily see this behaviour.
1. I'm running this commit from the main branch `e2588d6a42`
2. Slightly adjust your code by either:
  a. Checking out [this branch](https://github.com/fancom/zephyr/tree/spi123-incorrect-clock)
  b. Applying the patch `nucleo-configuration-patch` supplied below.
    - [Optional] Add `print-clocks-output` patch to print out debugging clock information
4. `west build -p auto -b nucleo_h743zi samples/drivers/led_ws2812/`
5. `west flash` onto a `nucleo_h743zi`
6. Probe `CN7:10` (`D13` or `SPI_A_SCK` or `SPI_1_SCK`) with an oscilloscope
7. Witness the `SCK` frequency incorrectly being 6MHz instead of the desired and calculated 3MHz.

It will also output the following with the optional logging patch 
```
[00:00:00.005,000] <err> spi_ll_stm32: Is this SPI1? 1
[00:00:00.005,000] <err> spi_ll_stm32: spi clock bus: STM32_CLOCK_BUS_APB2
[00:00:00.005,000] <err> spi_ll_stm32: STM32_CLOCK_BUS_APB2 freq: 96000000
[00:00:00.005,000] <err> spi_ll_stm32: LL_RCC_SPI123_CLKSOURCE: LL_RCC_SPI123_CLKSOURCE_PLL1Q
[00:00:00.005,000] <err> spi_ll_stm32: pllq freq: 192000000
[00:00:00.005,000] <err> spi_ll_stm32: Desired frequency: 3000000
[00:00:00.005,000] <err> spi_ll_stm32: BR: 5 should create clock of freq: 3000000
[00:00:00.005,000] <err> spi_ll_stm32: Resulting actual clock clock with PLL: 6000000
````

## Expected behavior
Expect the prescaler calculation to use the correct clock source to output the desired frequency.

## Impact
Hard to notice issues with incorrect SPI frequencies meaning devices run slower/faster than designed for.

## Environment (please complete the following information):
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.2
 - Commit SHA: `e2588d6a42`

## Additional Context
I created #40998 to attempt to fix this issue, but ABOSTM found a showstopping flaw in my implementation. We then proposed a new sollution would be to:

- Use the default `spi_ker_ck` value, since it's not possible to set two clocks the DTS. This means SCK source will always be `PLL1Q`.
- When calculating the prescaler use PLL1Q clock value.
- Add checks to ensure `PPL1Q` is enabled and `#error` if not.

I will try to create a fix for this implementation soon. However, I'm not sure yet what would be the best way to calculate the `PLL1Q` frequency, and would like some thoughts on that. The way I did it in the patch below is quite ugly.

It's coincidance that this has gone unnoticed during previous SPI driver testing. If I understand correctly then the Nucleo H743zi is popular for STM32H7 testing, and the `PLL1 Q` divider usually sets the clock up by chance that `PLL1Q == APB2`. Therefore I had to change the dts in the patch below to change `pll1_q` divisor from `4` to `1` to highlight this issue.

## Patches

### **nucleo-configuration-patch**
```
From 21c48a1f7ec70e773cf7639c3d0a3175ab390ff3 Mon Sep 17 00:00:00 2001
From: Hein Wessels <hein.wessels@nobleo.nl>
Date: Fri, 7 Jan 2022 14:19:30 +0100
Subject: [PATCH] nucleo config

---
 .../led_ws2812/boards/nucleo_h743.conf        |  2 +
 .../led_ws2812/boards/nucleo_h743zi.overlay   | 39 +++++++++++++++++++
 2 files changed, 41 insertions(+)
 create mode 100644 samples/drivers/led_ws2812/boards/nucleo_h743.conf
 create mode 100644 samples/drivers/led_ws2812/boards/nucleo_h743zi.overlay
[... 168 lines cut ...]
+			break;
+		case LL_RCC_SPI123_CLKSOURCE_CLKP:
+			LOG_ERR("LL_RCC_SPI123_CLKSOURCE: %s", "LL_RCC_SPI123_CLKSOURCE_CLKP");
+			break;			
+	}	
+	LOG_ERR("pllq freq: %d", get_pllq_frequency());
+
+	LOG_ERR("Desired frequency: %d", config->frequency);
+
 	for (br = 1 ; br <= ARRAY_SIZE(scaler) ; ++br) {
 		uint32_t clk = clock >> br;
 
 		if (clk <= config->frequency) {
+			LOG_ERR("BR: %d should create clock of freq: %d", br, clk);
 			break;
 		}
 	}
+	
+	LOG_ERR("Resulting actual clock clock with PLL: %d", get_pllq_frequency() >> br);	
 
 	if (br > ARRAY_SIZE(scaler)) {
 		LOG_ERR("Unsupported frequency %uHz, max %uHz, min %uHz",
-- 
2.25.1
```

erwango, 2022-01-07:
Thanks for raising this @heinwessels.

I'm currently looking at supporting multiple/alternate peripheral clock source by STM32 clock_control driver.
One of the advantage would be to centralize most of the work in one common driver, rather than splitting
it in multiple peripheral drivers (which would also have to support similar problem for multiple series)

I still need to give it more thoughts before sharing more details but it looks like the case you mention would be a good starting point.

Btw, and this is a different topic but I see that the sample you're using is `led_strip: ws2812`.
This is worrying me a little since we have identified a different issue with this driver, along with a different solution.
Would you mind having a look to following points ?
- https://github.com/zephyrproject-rtos/zephyr/issues/38683 (identified issue)
- https://github.com/zephyrproject-rtos/zephyr/pull/40860 (proposed solution)
I'd like to be sure this is orthogonal to the problem you're seeing on H7 SPI clocks.

heinwessels, 2022-01-07:
> Btw, and this is a different topic but I see that the sample you're using is `led_strip: ws2812`. This is worrying me a little since we have identified a different issue with this driver, along with a different solution. Would you mind having a look to following points ?

@erwango I do believe this not at all a framing issue, and therefore not related to the issue you mentioned. It's also not limited to the `ws2812` at all. I picked up this issue when using the SPI for normal communiation (controlling a `ili9341` LCD display). I simply used `ws2812` sample as it seemed like the fastest way to get a dummy SPI peripheral running as minimal working example.

> I'm currently looking at supporting multiple/alternate peripheral clock source by STM32 clock_control driver.
One of the advantage would be to centralize most of the work in one common driver, rather than splitting
it in multiple peripheral drivers (which would also have to support similar problem for multiple series)

Do you think I should still do an temporary fix for this SPI case in the mean time? A more comprehensive clock driver is the better solution, but it might still be some while off. There might be users running at different SPI frequencies without knowing it until then.

erwango, 2022-01-07:
> @erwango I do believe this not at all a framing issue, and therefore not related to the issue you mentioned. It's also not limited to the ws2812 at all. I picked up this issue when using the SPI for [...]

Ok. I didn't want to say this was the root cause of the clock issue you've identified, as it seems to be a well established issue.
I just wanted to be sure we're not colliding 2 different issues here by picking up an already broken example. And if not the case then I wonder if #40860 is required if you're able to get the sample working w/o it.
Though, anyway as mentioned, this is a totally different topic.

> Do you think I should still do an temporary fix for this SPI case in the mean time? 

One idea would be to push it as a PR and not getting it merged, so it is available to people that may face the issue.
It may also be a good comparison point to the upcoming clock_control driver enhancement.
I'm reluctant to merge this code if we already know it has to be removed rather soon, but in case of known delay, it is still an option.

heinwessels, 2022-01-07:
> I just wanted to be sure we're not colliding 2 different issues here by picking up an already broken example. 

I don't think that's the case :) I made really sure that it's indeed a bug, because it's been giving me quite a few headaches.

> I'm reluctant to merge this code if we already know it has to be removed rather soon, but in case of known delay, it is still an option.

I understand completely. Then it will be lower priority for me to create a temporary fix. In our implementation I will simply change the `PLL Q` divisor so that it matches the expected clock source frequency for now, since it's not used for anything else. Hacky, but should suffice until your refactoring at some point.

PR #45053 (2022-05-10): [RFC] stm32: clock_control: Enable device clock source selection
## TL;DR ##
STM32 clock_control driver update to provide a way to configure alternate clocks on peripherals that support it:
```
    clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00001000>,  <- Existing, bus driven, "register" clock
             <&rcc STM32_SRC_PLL3_P SPI123_SEL(2)>;   <- Newly added alternate clock (PLL3_P output here)
```
This alternate clock source could be selected in peripheral driver thanks to new `clock_control_configure()` API:
```
	if (IS_ENABLED(STM32_SPI_OPT_CLOCK_SUPPORT) && (cfg->pclk_len > 1)) {
		if (clock [... cut]
commit 0b65c1c6 (2022-05-10): drivers/spi: stm32: Use alt clock freq if available

Add support for an alternate clock. If available,
alternate clock is enabled and used to get the
device clock rate.

Fixes #41650
closed 2022-05-10 by carlescufi