zephyrproject-rtos/zephyr #41662: DMA Support for ATSAME54-XPRO

preetham05, 2022-01-07:
A communication via UART between STM32WB board and ATSAME54 board is desired. UART communication with DMA would be faster and efficient but ATSAME54 does not have DMA enabled. Adding DMA feature to this board would be helpful for my project.

erwango, 2022-03-03:
@FRASTM Could you provide a sample on how to use DMA on UART on WB boards?

nandojve, 2022-03-03:
Hi @preetham05 ,

SAME54 have DMA support. Please, check [tests/drivers/uart/uart_async_api/boards/atsame54_xpro.overlay](https://github.com/zephyrproject-rtos/zephyr/blob/eba55eacee7b1e94d60a5aa778bd892fb5bd64ca/tests/drivers/uart/uart_async_api/boards/atsame54_xpro.overlay)

Notes:

 * Once you enable DMA/Async API you will need to define all DMA channels for each SERCOM in UART mode. You can check this as an example [tests/drivers/uart/uart_async_api/boards/atsamr21_xpro.overlay](https://github.com/zephyrproject-rtos/zephyr/blob/afb23b2b56d399699177b0f584211bd7a96020e1/tests/drivers/uart/uart_async_api/boards/atsamr21_xpro.overlay)
 * Test configure UART in loopback mode. Make sure you configure it properly for your application.