zephyrproject-rtos/zephyr #41665: device_get_binding() unnecessarily inefficient if device is not ready

palchak-google, 2022-01-07:
**Is your enhancement proposal related to a problem? Please describe.**

The `device_get_binding()` performs a multi-pass linear search over all of the devices in the system to find a particular device by name. The first pass compares pointers to look for the name. The second pass does a true `strcmp` on the names.

If the desired device is found partway through one of the passes, but the device is not ready, the search algorithm will continue searching. This means that if a device is not active, the algorithm will *always* run both passes to completion and before returning NULL.

The body of the algorithm can be seen here:
[https://github.com/zephyrproject-rtos/zephyr/blob/35e8f42bc1b0d7f7cdaf272d1dbf8a36de4b5c40/kernel/device.c#L108](https://github.com/zephyrproject-rtos/zephyr/blob/35e8f42bc1b0d7f7cdaf272d1dbf8a36de4b5c40/kernel/device.c#L108)

**Describe the solution you'd like**
The algorithm should always return as soon as it's found a device with the correct name. The return value should indicate whether the device is ready.

This would look something like:
```
	for (dev = __device_start; dev != __device_end; dev++) {
		if (dev->name == name) {
			return z_device_is_ready(dev) ? dev : null;
		}
	}

	for (dev = __device_start; dev != __device_end; dev++) {
		if (strcmp(name, dev->name) == 0) {
			return z_device_is_ready(dev) ? dev : null;
		}
	}
```

tbursztyka, 2022-01-19:
Please send a patch, sounds like a bug to me.

rruuaanng, 2024-11-22:
It's changed. I'll close this issue :)

```
	for (dev = __device_start; dev != __device_end; dev++) {
		if (z_device_is_ready(dev) && (dev->name == name)) {
			return dev;
		}
	}
```

rettichschnidi, 2024-11-23:
Yes, the code got changed. But the proposed change has not been implemented.

rruuaanng, 2024-11-24:
I confirmed the behavior, it does a secondary search when it finds a device that is not prepared but exists, which is meaningless. I apologize for closing this issue, I will issue a PR to fix it :).
(There is actually only one solution, as shown above.)

kartben, 2024-11-27:
Fixed with #81833

closed 2024-11-27 by kartben