zephyrproject-rtos/zephyr #41677: undefined reference to `__device_dts_ord_xx'

beiszhihao, 2022-01-10:
**Describe the bug**
Hi zephyr developers, Hello!

I recently developed a wm89xx coder driver for a development board. My example is drivertree. The registration code is as follows:
```c
#define WM89XX_CONFIG_I2C(inst) \
        {                       \
                .bus.i2c = I2C_DT_SPEC_INST_GET(inst), \
                .bus_io = &wm89xx_bus_io_i2c,           \
        }
#define WM89XX_DEFINE(inst)\
        static struct WM89XX_Data wm89xx_data_##inst; \
        static const struct WM89XX_Config wm89xx_config_##inst = \
                                                WM89XX_CONFIG_I2C(inst); \
        DEVICE_DT_INST_DEFINE(inst,     \
                              wm89xx_init,\
                              NULL,\
                              &wm89xx_data_##inst,\
                              &wm89xx_config_##inst,\
                              POST_KERNEL,\
                              MY_INIT_PRIO,\
                              &wm89xx_api);

DT_INST_FOREACH_STATUS_OKAY(WM89XX_DEFINE)
```
But I had a problem compiling:
```bash
:/home/zhihao/zephyrproject/zephyr/drivers/wm89xx/src/wm89xx.c:39: undefined reference to `__device_dts_ord_92'
collect2: error: ld returned 1 exit status
```
Remind me undefined reference to `__device_dts_ord_92'，I studied it carefully. The problem lies in this registration macro
```c
DEVICE_DT_INST_DEFINE(inst,     \
                              wm89xx_init,\
                              NULL,\
                              &wm89xx_data_##inst,\
                              &wm89xx_config_##inst,\
                              POST_KERNEL,\
                              MY_INIT_PRIO,\
                              &wm89xx_api);
```
So I went to check the implementation of this macro in the zephyr source code and tried to get some useful information from it
```c
/**
 * @def DEVICE_DT_INST_DEFINE
 *
 * @brief Like DEVICE_DT_DEFINE for an instance of a DT_DRV_COMPAT compatible
 *
 * @param inst instance number. This is replaced by
 * <tt>DT_DRV_COMPAT(inst)</tt> in the call to DEVICE_DT_DEFINE.
 *
 * @param ... other parameters as expected by DEVICE_DT_DEFINE.
 */
#define DEVICE_DT_INST_DEFINE(inst, ...) \
        DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
```
Through the above definition and description, I probably understand__ device_ dts_ ord_ 92, of which 92 is the ID of the device in the tree, so I think this function is an automatically generated function, so theoretically it should be automatically generated by the zephyr construction tool, but I don't know why this function is not generated for my driver project, so I checked zephyr map
```c
zephyr.map:2282:                0x0000000008003b60                __device_dts_ord_18
zephyr.map:2286:                0x0000000008003b78                __device_dts_ord_274
zephyr.map:2287:                0x0000000008003b90                __device_dts_ord_273
zephyr.map:2288:                0x0000000008003ba8                __device_dts_ord_19
zephyr.map:2289:                0x0000000008003bc0                __device_dts_ord_272
zephyr.map:2290:                0x0000000008003bd8                __device_dts_ord_271
zephyr.map:2291:                0x0000000008003bf0                __device_dts_ord_270
zephyr.map:2292:                0x0000000008003c08                __device_dts_ord_269
zephyr.map:2293:                0x0000000008003c20                __device_dts_ord_268
zephyr.map:2294:                0x0000000008003c38                __device_dts_ord_49
zephyr.map:2295:                0x0000000008003c50                __device_dts_ord_267
zephyr.map:2296:                0x0000000008003c68                __device_dts_ord_69
[... 77 lines cut ...]
config LOG
        default y

config LOG_PRINTK
        default y

config SENSOR_LOG_LEVEL
        default 4

# Enable SPI and I2C support by default so that the sample works with
# the device connected either way. These defaults can be overridden if
# needed.
config SPI
        default y

config I2C
        default y

source "Kconfig.zephyr"
```
I began to try it. I found that when I set `config I2C` to N, the same problem as me will appear, which surprised me very much, because this problem has made progress. When I set I2C to y in my driver kconfig, my project was surprisingly compiled and passed, and there will be no previous problems.
So I guess the reason is `Z_DEVICE_DEFINE`  The macro define gets the properties of the dependent device node, where it uses `COND_ CODE_ 1` this macro performs conditional compilation. If the dependent node function has no driver, zephyr will not compile the I2C device. If the dependent device driver has not been compiled, zephyr will not generate it`__ device_ dts_ ord` function

In order to further verify my conjecture, my `Prj.conf `in the app project Add config to `CONF_I2C=y`, it is found that the effect is the same as that in `kconfig`, so my conclusion is basically determined. The above is my debugging process and solution to this problem.
When this problem occurred, I tried to search for information on the Internet and found that someone also encountered this problem, but there was no solution, so I Xia

mbolivar-nordic, 2022-01-13:
Hi there @beiszhihao ! Thanks for the detailed report. I need to confirm 2 things with you.

1. Do you have this line of code:

```c
#define DT_DRV_COMPAT micro_wm89xx
```

In the same file you are using `DT_INST_FOREACH_STATUS_OKAY`? You must define `DT_DRV_COMPAT` for this to work. 

Please see this documentation for details and examples for how to use `DT_INST_FOREACH_STATUS_OKAY`: https://docs.zephyrproject.org/latest/guides/dts/howtos.html#option-1-create-devices-using-instance-numbers

2. Do you have a devicetree binding written for the `"micro,wm89xx"` compatible? See here for more information about bindings: https://docs.zephyrproject.org/latest/guides/dts/bindings.html#dt-bindings

ceolin, 2022-01-29:
I saw this  happen in a different scenario happens with power domain. In the follow example you have a domain and two devices using it. The DT will generate aa dependency between them, but the domain may not be defined if we don't build with this feature enabled. In this situation the same problem will happen, we will have an undefined reference  ...
```
/ {
	test_domain: test_domain {
		compatible = "power-domain";
		status = "okay";
	};

	test_dev_a: test_dev_a {
		compatible = "test-device-pm";
		status = "okay";
		power-domain = <&test_domain>;
	};

	test_dev_b: test_dev_b {
		compatible = "test-device-pm";
		status = "okay";
		power-domain = <&test_domain>;
	};
};
```

So, if I understood correctly, we can't have a device "depending" on other that is not defined, even if this dependency is optional.

mbolivar-nordic, 2022-02-11:
@beiszhihao I know you have given me a detailed report, but I think there may be some missing piece of information. If you are still having problems, could you please provide a git branch that reproduces the issue, along with a `west build` command that shows how it happens? The `undefined reference to __device...` is a generic problem that happens if you try to get a device pointer from a devicetree node if no such device exists. The underlying infrastructure is well tested so I suspect there is a misconfiguration in the driver itself.

jeremyherbert, 2022-02-21:
hi @mbolivar-nordic I am having this exact same problem. I have created a repo [here](https://github.com/jeremyherbert/zephyr_testrepo) which just implements a simple fake accelerometer sensor driver, if you just clone and then `west build` (board is set inside the `CMakeLists.txt` file) you will see the error: 

`src/main.c:30: undefined reference to `__device_dts_ord_3'`

I am unsure if the cause is related, but if you could point me in the right direction for how to debug these DTS binding issues it would be greatly appreciated.

jeremyherbert, 2022-02-21:
I have solved my error. There was a spelling mistake in the DRV_COMPAT declaration for my custom driver (although to be fair, this isn't exactly a helpful error message for someone who doesn't deeply understand the zephyr build system)

Shivam1282, 2022-03-03:
@mbolivar-nordic I tried all the above steps but I'm still not able to compile the code. I am compiling the lvgl in the subsys directory of samples in zephyr for stm32h7b3i_dk board.

[image]

closed 2022-03-08 by carlescufi

josuah, 2024-01-27:
No issue with this, but for whoever finding this issue from a search engine, here is the Zephyr documentation about it:
https://docs.zephyrproject.org/latest/build/dts/troubleshooting.html
Thanks!

pkoetter, 2024-05-08:
Thanks for your input!
I had a similar problem after using the example-application as a blueprint for my own project. However the issue I found that in the file <your_repository_with_west.yml>/zephyr/module.yml (see [example](https://github.com/zephyrproject-rtos/example-application/blob/main/zephyr/module.yml)) the line `dts_root: .` was previously commented by me. Uncommenting it probably made the <your_repository_with_west.yml>/dts/bindings folder visible to the build system.

gjhave, 2024-09-29:
Nice share, I got same issue, but cause by different mistake, share with you:
      compatible: "sc7a20-sensor"
      #define DT_DRV_COMPAT sc7a20-sensor
      #define INST_DT_SC7A20_SENSOR(inst) DT_INST(inst, sc7a20-sensor)

change sc7a20-sensor to sc7a20_sensor and the problem will be solved.
The reason is "-" is illegal in C/C++