zephyrproject-rtos/zephyr #41682: ESP32 mcuboot

sam131208, 2022-01-10:
**Describe the bug**
When I follow the doc instruction on https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html, it failed to build the mcuboot.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b esp32 -d build_mcuboot bootloader/mcuboot/boot/zephyr

**Expected behavior**
OTA the application by BLE.

**Logs and console output**
[2/26] Building C object CMakeFiles/app.dir/flash_map_extended.c.obj
FAILED: CMakeFiles/app.dir/flash_map_extended.c.obj
C:\Users\Sam\.espressif\tools\zephyr\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DBUILD_VERSION=v2.7.99-2863-gf882d43b56e9 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -IE:/zephyrrtos/bootloader/mcuboot/boot/zephyr/include -IE:/zephyrrtos/bootloader/mcuboot/boot/zephyr/targets -IE:/zephyrrtos/bootloader/mcuboot/boot/bootutil/include -IE:/zephyrrtos/zephyr/include -IE:/zephyrrtos/build_mcuboot/zephyr/include/generated -IE:/zephyrrtos/zephyr/soc/xtensa/esp32 -IE:/zephyrrtos/zephyr/soc/xtensa/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/include/bt -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/include/crypto -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../esp_shared/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../esp_shared/include/wifi -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/hal/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/hal/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32 -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32/private_include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_common/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/xtensa/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/xtensa/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/soc/include/soc -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/soc/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/soc/src/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/driver/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/efuse/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/efuse/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_system/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/esp32/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/private_include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_netif/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/esp_event/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/log/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/port/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include/esp_supplicant -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src/crypto -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/private_include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/newlib/priv_include -IE:/zephyrrtos/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include_bootloader -IE:/zephyrrtos/bootloader/mcuboot/boot/bootutil/zephyr/.. -IE:/zephyrrtos/bootloader/mcuboot/boot/bootutil/zephyr/../include -IE:/zephyrrtos/bootloader/mcuboot/boot/bootutil/zephyr/../../zephyr/include -IE:/zephyrrtos/modules/crypto/mbedtls/include -IE:/zephyrrtos/zephyr/modules/mbedtls/configs -isystem E:/zephyrrtos/zephyr/lib/libc/minimal/include -Og -imacros E:/zephyrrtos/build_mcuboot/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -imacros E:/zephyrrtos/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=E:/zephyrrtos/bootloader/mcuboot/boot/zephyr=CMAKE_SOURCE_DIR -fmacro-prefix-map=E:/zephyrrtos/zephyr=ZEPHYR_BASE -fmacro-prefix-map=E:/zephyrrtos=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -Wno-unused-variable -Wno-maybe-uninitialized -std=c99 -MD -MT CMakeFiles/app.dir/flash_map_extended.c.obj -MF CMakeFiles\app.dir\flash_map_extended.c.obj.d -o CMakeFiles/app.dir/flash_map_extended.c.obj -c E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c
In file included from E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:11:
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/include/target.h:42:2: error: #error "Target support is incomplete; cannot build mcuboot."
 #error "Target support is incomplete; cannot build mcuboot."
  ^~~~~
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:27:2: error: #error "FLASH_DEVICE_ID could not be determined"
 #error "FLASH_DEVICE_ID could not be determined"
  ^~~~~
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c: In function 'flash_device_base':
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:42:18: error: 'FLASH_DEVICE_ID' undeclared (first use in this function); did you mean 'FLASH_AREA_ID'?
     if (fd_id != FLASH_DEVICE_ID) {
                  ^~~~~~~~~~~~~~~
                  FLASH_AREA_ID
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:42:18: note: each undeclared identifier is reported only once for each function it appears in
In file included from E:/zephyrrtos/zephyr/include/toolchain.h:50,
                 from E:/zephyrrtos/zephyr/include/kernel_includes.h:19,
                 from E:/zephyrrtos/zephyr/include/kernel.h:17,
                 from E:/zephyrrtos/zephyr/include/zephyr.h:18,
                 from E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:8:
E:/zephyrrtos/zephyr/include/sys/cbprintf_internal.h:266:15: error: expression in static assertion is not an integer
  BUILD_ASSERT(!((sizeof(double) < VA_STACK_ALIGN(long double)) && \
               ^
E:/zephyrrtos/zephyr/include/toolchain/gcc.h:61:51: note: in definition of macro 'BUILD_ASSERT'
 #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
                                                   ^~~~
E:/zephyrrtos/zephyr/include/sys/cbprintf_internal.h:292:2: note: in expansion of macro 'Z_CBPRINTF_PACK_ARG2'
  Z_CBPRINTF_PACK_ARG2(_pbuf, _pkg_len, _pkg_offset, _pmax, _flags, \
  ^~~~~~~~~~~~~~~~~~~~
E:/zephyrrtos/zephyr/include/sys/util_loops.h:1075:2: note: in expansion of macro 'Z_CBPRINTF_PACK_ARG'
  fixed_arg0(x)
  ^~~~~~~~~~
E:/zephyrrtos/zephyr/include/sys/util_loops.h:38:2: note: in expansion of macro 'Z_FOR_EACH_EXEC'
  z_call(2, x, fixed_arg0, fixed_arg1)
  ^~~~~~
E:/zephyrrtos/zephyr/include/sys/util_loops.h:23:53: note: in expansion of macro 'Z_FOR_LOOP_3'
     _57, _58, _59, _60, _61, _62, _63, _64, N, ...) N
                                                     ^
[... cut ...]
In file included from E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/main.c:29:
E:/zephyrrtos/bootloader/mcuboot/boot/zephyr/include/target.h:42:2: error: #error "Target support is incomplete; cannot build mcuboot."
 #error "Target support is incomplete; cannot build mcuboot."
  ^~~~~
[7/26] Building C object CMakeFiles/app.dir/E_/zephyrrtos/bootloader/mcuboot/boot/bootutil/src/loader.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'E:\zephyrrtos\build_mcuboot'

sylvioalves, 2022-01-10:
@sam131208, MCUBoot porting for ESP32 SOC is still a standalone procedure, which means you can't yet build as the sample describes. Please, follow the documentation below to build and flash the mcuboot bootloader. Once flashed, you need only to work on the sample application.  
https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md

dkalowsk, 2022-01-11:
Closing as per @sylvioalves comment

closed 2022-01-11 by dkalowsk

sam131208, 2022-01-13:
@sylvioalves Thanks! it is success on Ubuntu build the mcuboot bootloader while failed on windows with Python310.
    But what confuses me is, how set the BOOTLOADER_FLASH_OFFSET while Flash MCUboot. Whether it is set BOOTLOADER_FLASH_OFFSET=0x1000, or BOOTLOADER_FLASH_OFFSET=0x10000, it can be successfully bootload  application.

sam131208, 2022-01-13:
@sylvioalves While  enable the mcuboot, the Bluetooth initial is failed.
west build -b esp32 samples/bluetooth/peripheral/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y
west flash --bin-file build/zephyr/zephyr.signed.confirmed.bin

 =~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.01.13 11:34:27 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4604
ho 0 tail 12 room 4
load:0x40085000,len:14964
load:0x4009d000,len:3624
entry 0x40085478
[esp32] [INF] *** Booting MCUboot build v1.7.0-421-g399720d ***
[esp32] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] DRAM segment: start=0x54e8, size=0x17e4, vaddr=0x3ffbdb60
[esp32] [INF] IRAM segment: start=0x6ccc, size=0x115d0, vaddr=0x40080000
[esp32] [INF] start=0x4009014c
*** Booting Zephyr OS build v2.7.99-3070-g7eb50bebe486  ***
Bluetooth init failed (err -45)
[00:00:01.119,000] <err> bt_settings: settings_subsys_init failed (err -45)