zephyrproject-rtos/zephyr #41683: http_client: Unreliable rsp->body_start pointer

mglettig, 2022-01-10:
**Describe the bug**
Starting with Zephyr v2.7 I observed that rsp->body_start information in the http_client callback is not always reliable. Sometimes this pointer is set in the middle of a large reply.

- The target platform that I'm using is NRF9160.
- A potential workaround is to only use the rsp->body_start pointer once a the beginning of a server reply and latter ignore this pointer.

**To Reproduce**
Steps to reproduce the behavior:
I observed this issue in my custom application. It should be possible to reproduce the issue using a modified version of https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/sockets/http_client when you download big files. In my case the downloaded file was ~8MB and was downloaded over HTTPS.

**Expected behavior**
I would expect the rsp->body_start is either set to the correct address or is NULL if invalid.

**Impact**
It's not a showstopper but it's annoying. I used a lot of time to track down this issue.

**Environment (please complete the following information):**
 - Zephyr v2.7

ycsin, 2022-01-21:
I think this [line](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/http/http_client.c#L268) might be the root of confusion?

I reckon that it is reasonable for a user of the `http_client` to expect that the `body_start` pointer always points to the address of the body, whenever there's body in the http response, but instead the `body_start` is pointed to the body only when the response contains header + body. If the response contains body only, the `body_start` will be NULL & it is required to get the body by `recv_buf` instead:

[image]

I'd also suggest the `http_client` to calculate things like body length internally, so that user can deal with the data directly without manually deriving the pointers to the header/body and calculating their respective length in the application.

rlubos, 2022-01-21:
@ycsin I agree with you, I was actually investigating this issue yesterday, and concluded that the `body_start` pointer should be set  unconditionally, regardless of the position in the buffer, otherwise the behaviour is unreliable. I'll post a PR shortly.

PR #42026 (2022-01-21): net: http_client: Set body_start pointer unconditionally
Set `body_start` pointer regardless of the body position in the recv
buffer. In result, the pointer shall indicate correctly position of the
body for each fragment, it's also explicit now that if the pointer is
not set for a fragment, there's no body in that particular fragment.

Fixes #41683
closed 2022-01-21 by nashif