zephyrproject-rtos/zephyr #41686: SPI CS signal not used in SSD1306 driver

rohand87, 2022-01-10:
**Describe the bug**

I am trying to run the `drivers/display` sample code on a OLED display having SS1306 driver in 4 wire SPI mode. My host MCU is STM32G474RET. The code builds and runs properly but nothing happens on the display. I am sure the hardware is ok because I have tested [this code](https://github.com/afiskon/stm32-ssd1306) on the same hardware. 

After some debugging I found that the SSD1306 driver does not make use of the CS signal. I believe this is the issue. Can someone please confirm if they have successfully run the same sample code on an ssd1306 based oled display in SPI mode? 

Below is my dtsi file for the spi interface
```
&spi2 {
    status = "okay";
    pinctrl-0 = <&spi2_sck_pf1 &spi2_mosi_pb15 &spi2_nss_pf0>;
    pinctrl-names = "default";

    ssd1306: ssd1306@0 {
		compatible = "solomon,ssd1306fb";
		reg = <0x0>;
		label = "OLED_SSD1306";
		spi-max-frequency = <8000000>;
		width = <128>;
		height = <64>;
		segment-offset = <0>;
		page-offset = <0>;
		display-offset = <0>;
		multiplex-ratio = <63>;
		segment-remap;
		/*com-invdir;*/
		prechargep = <0x22>;
		data_cmd-gpios = <&gpiob 9 GPIO_ACTIVE_LOW>;
		reset-gpios = <&gpiob 7 GPIO_ACTIVE_LOW>;
	};
};
```

Kconfigs used
```
CONFIG_DISPLAY=y
CONFIG_SSD1306=y
```

rivimey, 2022-01-16:
Roland, I had a look through the code, and one thing worth checking is whether, in your board stack, the Device tree actually defines a pin for CS. If it doesn't the driver can't use it.

If it does and the driver doesn't reference it, that is surely a bug, but would in theory be true for many users of SPI as the SSD driver uses generic SPI calls.

rohand87, 2022-01-19:
Hi Rivimey,
The issue was in my device tree `data_cmd-gpios = <&gpiob 9 GPIO_ACTIVE_LOW>;` line. I changed it to `data_cmd-gpios = <&gpiob 9 GPIO_ACTIVE_HIGH>;` and the code started working. I believe the CS pin wasn't the issue. Thanks for your reply.

erwango, 2022-01-19:
@rohand87 If the issue is fixed and linked to your configuration, can you close the issue ?

closed 2022-01-19 by rohand87