zephyrproject-rtos/zephyr #41694: undefined reference to `_open'

iparobin, 2022-01-10:
Hello,
I don't know if I missed something but I'm trying to use the POSIX support for thread management for the nrf52833 board. I made this choice to be compatible with other OS.
I have integrated the following config to enable POSIX in prj.conf:

```
CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
CONFIG_POSIX_API=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP11=n
```

A part of my CMakelist for the compilation of my module :
```
  zephyr_interface_library_named(${PROJECT_NAME})
  zephyr_include_directories( ${PROJECT_NAME} INTERFACE include/  )
                      
  zephyr_library_sources(src/test.cpp)

  zephyr_library_link_libraries(${PROJECT_NAME})
  target_link_libraries(${PROJECT_NAME} INTERFACE zephyr_interface test2)
```

And I have the following problem whenever I call pthread_create:
```
[163/173] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /media/dev/zephyr/zephyr_products/build/zephyr/zephyr_pre0.map 
: && ccache /home/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/media/dev/zephyr/zephyr_products/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/media/dev/zephyr/zephyr_products/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lm  -Wl,-lc  -L"/home/zephyr-sdk-0.13.1/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m/nofp  -u_printf_float  -Wl,-lgcc  -lc  -specs=nano.specs && cd /media/dev/zephyr/zephyr_products/build/zephyr && /home/.local/lib/python3.6/site-packages/cmake/data/bin/cmake -E echo
/home/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/zephyr-sdk-0.13.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libc_nano.a(lib_a-openr.o): in function `_open_r':
openr.c:(.text._open_r+0x10): undefined reference to `_open'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I compile with west.
Do you have an idea of the problem?

closed 2022-01-11 by iparobin

cfriedt, 2022-12-14:
Solution copied from discord

> Ah found it, had to set CONFIG_FILE_SYSTEM to get CONFIG_POSIX_FS enabled

brightproject, 2024-09-27:
Maybe this will help someone.
The reason is that somewhere in your code there is

> fopen();

The error points to just that.

> openr.c:(.text._open_r+0x10): undefined reference to `_open'