zephyrproject-rtos/zephyr #41708: Set compiler options only for the application

simon-marti-loepfe, 2022-01-11:
Is there a way to set the compiler options only for the application? I would like to enable `-Wall` ,`-Wextra` and `-Werror` but only for my application files. So far I could only find ways to set the compiler options globally. I tried the following to options:

a) Setting `CONFIG_COMPILER_OPT`. That changes the options for all files (as described in the [documentation](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_COMPILER_OPT.html)).
b) Setting `zephyr_library_compile_options` as described in issue #15065. That also affects all files.

It would be nice to have a configuration option like `CONFIG_COMPILER_OPT` that is only applied to the application files. In my case fixing all warnings that are generated when compiling all Zephyr code with `-Wextra` would take too much time, but I would still like to be able to enable the warnings for the application code.

campbellyoung, 2022-01-17:
This work for me:

```
target_compile_options(app PRIVATE -Wall -Werror -Wextra -Wno-unused-parameter -Wno-unused-function -Wno-ignored-qualifiers)
```

Then the only issue I have is that the warnings generated in headers (see below) aren't ignored because they're not using `-isystem`, but isn't a big deal for us. 

```
error: type qualifiers ignored on function return type [-Werror=ignored-qualifiers]
  568 | static inline const struct net_l2 * const net_if_l2(struct net_if *iface)
```

simon-marti-loepfe, 2022-01-17:
Thank you very much, setting `target_compile_options()` works. I also have to suppress the warnings created by the included headers (with `-Wno-unused-parameter -Wno-missing-field-initializers` in my case ), but that's acceptable.

closed 2022-01-17 by simon-marti-loepfe