zephyrproject-rtos/zephyr #41734: Can't enable pull-up resistors in ESP32 gpio 25,26,27

szczys, 2022-01-11:
**Describe the bug**

The pull-up resistors for GPIO25, GPIO26, and GPIO27 aren't being enabled and the pins are left floating. This behavior is shown on a dev board that has the ESP-WROOM-32 module as well as a TTGO T-Display board.

**To Reproduce**

```
/ {
    aliases {
        sw0 = &button0;
    };
    gpio_keys {
        compatible = "gpio-keys";
        button0: button_0 {
            gpios = <&gpio0 27 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
            label = "Button 0";
        };
    };
};
```

1. Create esp32.overlay file in the /basic/button/ folder of the [button sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/basic/button)
2. west build -b esp32 samples/basic/button/ -p auto
3. west flash
4. probe GPIO27 with a multimeter and its voltage will be floating (in the millivolt range)

**Expected behavior**

GPIO27 should be pulled up to 3.3V. If you change the overlay file to `&gpio0 21` and probe GPIO21 you will see the expected 3.3V on that pin (or any other lower-number pins).

The reference manual lists GPIO34..39 as not having pull-up/pull-down resistors. But GPIO25..27 should have them.

**Impact**
This is a gotcha that took a while to hunt down (some pins were working, some were not)

**Environment (please complete the following information):**
 - OS: Linux, MacOS
 - Zephyr SDK, West version: v0.12.0
 - 2bf02f4854aae9b936f4c3267699d2dc7ea16bd9

sylvioalves, 2022-01-19:
@szczys I think this was fixed already by the commit below:  
https://github.com/zephyrproject-rtos/zephyr/pull/40864/commits/370b3d8423bcbc7b760060480c3e1c6271827bd0

Would you update you main and re-test?  

Thanks!

szczys, 2022-01-19:
Yes, at the  current main (7dac931e36) I am able to use the pull-ups on these pins. Thank you for pointing this out and I apologize for not testing the newest before reporting.

closed 2022-01-19 by szczys

sylvioalves, 2022-01-19:
@szczys hey, no problem.