zephyrproject-rtos/zephyr #41736: Xtensa xt-xc++ Failed to build C++ code

hongshui3000, 2022-01-12:
**Describe the bug**
Failed to build C++ code
log
```
[100/155] Building CXX object CMakeFiles/app.dir/src/main.cpp.obj
FAILED: CMakeFiles/app.dir/src/main.cpp.obj 
C:\usr\xtensa\XtDevTools\install\tools\RI-2021.6-win32\XtensaTools\bin\xt-xc++.exe -DBUILD_VERSION=zephyr-v2.7.0 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -IE:/work/zephyr/include -IE:/work/x/build/zephyr/include/generated -IE:/work/x/soc/xtensa/x -IE:/work/x/soc/xtensa/x/include -IE:/work/x/include -IE:/work/x/modules/hal/xtensa/include -IE:/work/x/modules/hal/xtensa/zephyr/soc/x -isystem E:/work/x/zephyr/lib/libc/minimal/include -isystem E:/work/x/zephyr/subsys/cpp/include -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros E:/work/x/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacrosE:/work/x/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros E:/work/x/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -fgnu89-inline -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -nostdinc++ -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles\app.dir\src\main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c E:/work/x/samples/subsys/cpp/cpp_synchronization/src/main.cpp
xt-xc++.exe ERROR parsing -Wno-register:  unknown flag
[109/155] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
ninja: build stopped: subcommand failed.
```

dcpleung, 2022-01-21:
xc-x++ doesn't support C++11 so you will need to set `CONFIG_STD_CPP98=y` and `CONFIG_STD_CPP11=n` (basically enabling C++98 and disabling C++11).

dcpleung, 2022-01-25:
This is not a bug but a config issue. Closing...

closed 2022-01-25 by dcpleung