zephyrproject-rtos/zephyr #41742: stm32_temp: not working

ycsin, 2022-01-12:
I'm trying the stm32_temp driver on my G0B1RE custom board.

[image]

My device tree looks like this
```
/ {
	model = "GTSB IVM21";
	compatible = "gtsb,ivm21";

	chosen {
		zephyr,console = &usart2;
		zephyr,shell-uart = &usart2;
		zephyr,uart-mcumgr = &usart2;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &slot0_partition;
	};
...
	vmain {
		compatible = "voltage-divider";
		io-channels = <&adc1 0>;
		output-ohms = <51000>;
		full-ohms = <(51000 + 51000)>;
	};

	stm-temp {
		label = "DIE_TEMP";
		compatible = "st,stm32-temp";
		io-channels = <&adc1 12>;
		avgslope = <25>;
		v25 = <760>;
		status = "okay";
	};
};
...
&adc1 {
	pinctrl-0 = <&adc1_in0_pa0 &adc1_in1_pa1 &adc1_in2_pa2 &adc1_in3_pa3>;
	status = "okay";
};
```

And I added these into my defconfig
```
CONFIG_SENSOR=y
CONFIG_STM32_TEMP=y
```

with no changes to my existing application, I found that my application fails to read the battery voltage, the channel setup fails here everytime:

[image]

I then changed the stm32_temp driver [init](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/stm32_temp/stm32_temp.c#L104) to: (I'm not sure what am I doing, just trial and error)

```c
	*accp = (struct adc_channel_cfg){
		.gain = ADC_GAIN_1,
		.reference = ADC_REF_INTERNAL,
		.acquisition_time = ADC_ACQ_TIME_DEFAULT,
		.channel_id = data->channel,
		.differential = 0
	};
```

then I copy the stm32_temp sample code to my main() and found that the code stuck in the [calibration step](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L342), I then edited the stm32_temp driver so that the [calibration is set to false](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/stm32_temp/stm32_temp.c#L116).

```c
	*asp = (struct adc_sequence){
		.channels = BIT(data->channel),
		.buffer = &data->sample_buffer,
		.buffer_size = sizeof(data->sample_buffer),
		.resolution = 12,
		.calibrate = false,
	};
```

And now I'm able to read something, but it clearly doesn't work:
```
Current temperature: -172.-199996 °C
Current temperature: -172.-199996 °C
Current temperature: -171.-800003 °C
Current temperature: -171.-800003 °C
```

This is the characteristics from G0B1 datasheet:
[image]

I'm not sure if it has negative temperature coefficient, this is what I get if I enabled `ntc`:
```
Current temperature: 221.399993 °C
Current temperature: 222.199996 °C
Current temperature: 221.399993 °C
Current temperature: 222.199996 °C
```

I do know that the formula in stm32_temp isn't suppose to "just works", as it uses v25, and I think I need v30 (so probably need to change this [line](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/stm32_temp/stm32_temp.c#L76) to += 30?)

I'm on Zephyr 2.7.0 RC3, `CONFIG_PM` is enabled.

[15. ADC.pdf](https://github.com/zephyrproject-rtos/zephyr/files/7854188/15.ADC.pdf)
[stm32g0b1re.pdf](https://github.com/zephyrproject-rtos/zephyr/files/7854195/stm32g0b1re.pdf)

ycsin, 2022-01-12:
@Delsian do you have any idea?

erwango, 2022-01-12:
@ycsin G0 temp sensor probably requires this https://github.com/zephyrproject-rtos/zephyr/pull/41295

ycsin, 2022-01-13:
I think the first part of the problem (acquisition time) is due to this [line](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L789), where all channels need to have the same acquisition time on G0 series.

The second part (calibration) might be a bug in the adc driver which is fixed recently by bc1b262885c3e91b3391a10dc7ecf8b489065998

ycsin, 2022-03-22:
> The temperature is measured :
> 
> 
> ```
> Current temperature: 182.9 �°C                                                  
> Current temperature: 182.7 �°C                                                  
> Current temperature: 182.9 �°C                                                  
> Current temperature: 182.9 �°C  
> ```

Hmm.. 180+ degree celsius doesnt seem to be working to me

FRASTM, 2022-03-23:
based on the zephyr 1d3dbd49e14e9dec14a6bfa4d69fa56b62406784
I just add the following `board/nucleo_g071rb.overlay` to the samples/sensor/stm32_temp_sensor/
```
/ {
	stm-temp {
		label = "DIE_TEMP";
		compatible = "st,stm32-temp";
		io-channels = <&adc1 12>;
		status = "okay";
	};
};

&adc1 {
    status = "okay";
    has-temp-channel;
};
```

to get 
```
*** Booting Zephyr OS build zephyr-v3.0.0-1625-g1d3dbd49e14e  ***
STM32 Die temperature sensor test
Current temperature: 21.0 �°C
Current temperature: 20.6 �°C

```

FRASTM, 2022-03-23:
> > The temperature is measured :
> > ```
> > Current temperature: 182.9 �°C                                                  
> > Current temperature: 182.7 �°C                                                  
> > Current temperature: 182.9 �°C                                                  
> > Current temperature: 182.9 �°C  
> > ```
> 
> Hmm.. 180+ degree celsius doesnt seem to be working to me
> 
> EDIT1: I had a closer look at the #43861, I think this PR fixes the [`drivers/adc` sample](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/drivers/adc/src/main.c#L94) and anything that uses the `adc_ref_internal` API. As far as I can tell, neither `adc_ref_internal` nor `ref_internal` are used in the `stm32_temp` or the `adc_stm32`.
> 
> The `stm32_temp` driver also suffers the [same issue](https://github.com/zephyrproject-rtos/zephyr/pull/43863#discussion_r829753268) I mentioned in #43861.
> 
> EDIT2: Created PR #44052

value of 180°C    is due to a wrong 
		_avgslope = <43>;
		v25 = <1430>;_
default value have to be kept for stm32wb55 or stm32g0x1  (as defined by the Temperature sensor characteristics of the Datasheet)

FRASTM, 2022-03-23:
@ycsin can you please re-check on your side

ycsin, 2022-03-23:
@FRASTM I'm using the correct binding in my branch https://github.com/ycsin/zephyr/commit/1d58b73296f435234047f28263d010f43cc2f7a3 following the merge of https://github.com/zephyrproject-rtos/zephyr/pull/41295

I'd say the `stm32_temp` is working, but with a catch. It needs #44052 to work properly when more than one channel of the ADC is used.

closed 2022-04-29 by carlescufi