zephyrproject-rtos/zephyr #41754: Bluetooth, host: API to expose local controller version & manufacturer info to application

koffes, 2022-01-12:
Now one can print local controller and HCI info using BT_INFO, originating from https://github.com/zephyrproject-rtos/zephyr/blob/45a693a2a92412c8677cae9c3111bb4369272f42/subsys/bluetooth/host/hci_core.c#L3081

This is good, but it is desirable to have this information available to the application trough an API. 
Something similar to `bt_dev_get_info()` which returns `bt_dev` as a pointer, or it could be more specific, returning one of the 
```
hci_version;
lmp_version;
hci_revision;
lmp_subversion;
manufacturer;
```
Parameters.

koffes, 2022-11-01:
FYI @Emil-Gydesen-Bose

alwa-nordic, 2024-02-01:
Alternative: The app can run BT_HCI_OP_READ_LOCAL_VERSION_INFO.