zephyrproject-rtos/zephyr #41771: tests: drivers: adc: Test doesn't build for mec172xevb_assy6906

albertofloyd, 2022-01-13:
**Describe the bug**
ADC api driver test doesn't build for mec172xevb_assy6906

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/drivers/adc/adc_api
2. rm -rf build && west build -c  -b mec172xevb_assy6906
3. See error

**Expected behavior**
Test does build for this board

**Impact**
Limited test coverage for this board 

**Logs and console output**
/home/jamezaar/ecwork/zephyr_albertofloyd/tests/drivers/adc/adc_api/src/test_adc.c:332:2: error: #error "Unsupported board."
  332 | #error "Unsupported board."
      |  ^~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/tests/drivers/adc/adc_api/src/test_adc.c:347:22: error: 'ADC_GAIN' undeclared here (not in a function); did you mean 'ADC_GAIN_8'?
  347 |  .gain             = ADC_GAIN,
      |                      ^~~~~~~~
      |                      ADC_GAIN_8
/home/jamezaar/ecwork/zephyr_albertofloyd/tests/drivers/adc/adc_api/src/test_adc.c:348:22: error: 'ADC_REFERENCE' undeclared here (not in a function)
  348 |  .reference        = ADC_REFERENCE,

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2022-11-18:
no idea why this moved from bug to enhancement, without any comment explaining the reasoning. Move back to a bug, if the bug is invalid, close it.

carlescufi, 2022-11-23:
@nashif assigning to you since this has nothing to do with nRF

nashif, 2022-11-23:
> @nashif assigning to you since this has nothing to do with nRF

@carlescufi  This is assigned to @anangl  as the ADC maintainer, not because of nRF.

either way, can't re-produce and test builds fine for me.

albertofloyd, 2022-11-24:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/44733

closed 2022-11-24 by albertofloyd