zephyrproject-rtos/zephyr #41772: stm32: adc: Add support for VBAT internal channel

ycsin, 2022-01-13:
Related to #17375, https://github.com/zephyrproject-rtos/zephyr/issues/17375#issuecomment-1010870257:

> 2a96cc03478a348f301649870fa1d5b88005f70b added support for internal voltage reference source but left out the VBAT, I modified the code to include VBAT:
> ```c
>	if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_TEMPSENSOR) == channel_cfg->channel_id) {
>		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_TEMPSENSOR);
>	} else if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_VREFINT) == channel_cfg->channel_id) {
>		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VREFINT);
>	} else if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_VBAT) == channel_cfg->channel_id) {
>		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VBAT);
>	}
> ```
>
> But the ADC value I got doesn't match the voltage on the VBAT.

Zephyr 2.7.0 RC3

ycsin, 2022-01-13:
cc @erwango

erwango, 2022-01-13:
@ycsin Would you mind mentioning on which stm32 series you're working (description and title)?
(I actually think this should be G0, since you're ofte working on G0, but please consider not every body knows your habits)

ycsin, 2022-02-24:
@erwango I gave this another try today but still no luck, but I did find something in the adc driver that I think can be improved:

1. I guess this [line](https://github.com/zephyrproject-rtos/zephyr/blob/5ee306944b37fac3143f30e759478c6e8a9d36a4/drivers/adc/adc_stm32.c#L987) is not required in the init anymore, since we have [this](https://github.com/zephyrproject-rtos/zephyr/blob/5ee306944b37fac3143f30e759478c6e8a9d36a4/drivers/adc/adc_stm32.c#L852)?
2. ADC should be disabled before invoking `adc_stm32_set_common_path` for some of the series according to the LL_ADC files, but I don't seem to see this being done in the driver, I could be missing something though.
3. Shouldn't the common path be removed after the sampling is done?

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L987

erwango, 2022-02-24:
@FRASTM, maybe you can have a check ?

Also, please note that some work in on goi,g on that topic: https://github.com/zephyrproject-rtos/zephyr/pull/42567

ycsin, 2022-02-25:
cc @manoj153

FRASTM, 2022-03-15:
measuring VBAT on the stm32g0 requires a call to 
if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(LL_ADC_PATH_INTERNAL_VBAT) == channel_id) {
		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VBAT);
}

Then According to the ref man you mention "_As a consequence, the converted digital value is half the V BAT
voltage._"

ycsin, 2022-03-15:
@FRASTM thanks for looking into this. According to the wordings of the refman its half the voltage, but the schematics show "Vbat/3" so it might be one-third as well?

As mentioned in the OP, I've already did this
```c
adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VBAT);
```

prior to sampling the internal channel, but for some reason the value I got didn't match 1/2 or 1/3 of 3.3V.

affadli, 2022-03-15:
Hi @FRASTM , I also got similar issue once using VBAT ADC but I used STM32WB chip.
The ADC value that I got didnt match with battery.

Is there any proper example regarding ADC using VBAT channel?

ycsin, 2022-03-15:
> Is there any proper example regarding ADC using VBAT channel?

Nope, currently the upstream ADC driver doesnt support sampling VBAT.

FRASTM, 2022-03-15:
Investigating, it could  be a pb in the adc configuration when converting the 12b value to mv,
changing the VREF I get :
```
ADC reading: 1284 = 1034 mV                                                     
ADC reading: 1272 = 1024 mV                                                     
ADC reading: 1246 = 1003 mV                                                     

```

FRASTM, 2022-03-15:
With a simple stm32Cube application :  [ADC_MultiChannelSingleConversion](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/NUCLEO-G071RB/Examples_LL/ADC/ADC_MultiChannelSingleConversion)
 where the VBAT is included in the sample : we can see that connecting pin A2 (GPIO PA4) to VBAT
uhADCxConvertedData_VrefAnalog_mVolt = 3300 (aADCxConvertedData[0] = 4095)
but
uhADCxConvertedData_VBat_mVolt = 3639   (aADCxConvertedData[3] = 1375)

--> The VBAT measured externaly differs from the VBAT measured internally

FRASTM, 2022-03-15:
> Hi @FRASTM , I also got similar issue once using VBAT ADC but I used STM32WB chip. The ADC value that I got didnt match with battery.
> 
> Is there any proper example regarding ADC using VBAT channel?

With a simple stm32Cube application : [ADC_MultiChannelSingleConversion](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/NUCLEO-G071RB/Examples_LL/ADC/ADC_MultiChannelSingleConversion)
modified to get internal VBAT on channel 14, where the VBAT is included in the sample : we can see that connecting pin A2 (GPIO PA4) to VBAT : 

1.   add a fourth element in the array to get the Vbat :  `#define ADC_CONVERTED_DATA_BUFFER_SIZE   (   4U)`
2. ` __IO uint16_t uhADCxConvertedData_VBat_mVolt = 0U;  `
3.  `uhADCxConvertedData_VBat_mVolt = __LL_ADC_CALC_VREFANALOG_VOLTAGE(aADCxConvertedData[3], LL_ADC_RESOLUTION_12B);`
4. /* Set ADC measurement path to internal channels */
    `LL_ADC_SetCommonPathInternalCh(__LL_ADC_COMMON_INSTANCE(ADC1), (LL_ADC_PATH_INTERNAL_VREFINT | LL_ADC_PATH_INTERNAL_TEMPSENSOR | LL_ADC_PATH_INTERNAL_VBAT));`
5.   in the Configure_ADC 
    /* Set ADC group regular sequencer length and scan direction */
    `LL_ADC_REG_SetSequencerLength(ADC1, LL_ADC_REG_SEQ_SCAN_ENABLE_4RANKS);`
6. /* Set ADC group regular sequence: channel on the selected sequence rank. */

 ```
 LL_ADC_REG_SetSequencerRanks(ADC1, LL_ADC_REG_RANK_4, LL_ADC_CHANNEL_VBAT);
    /* Poll for ADC channel configuration ready */
    #if (USE_TIMEOUT == 1)
    Timeout = ADC_CHANNEL_CONF_RDY_TIMEOUT_MS;
    #endif /* USE_TIMEOUT */
    
    while (LL_ADC_IsActiveFlag_CCRDY(ADC1) == 0)
    {
    #if (USE_TIMEOUT == 1)
      /* Check Systick counter flag to decrement the time-out value */
      if (LL_SYSTICK_IsActiveCounterFlag())
      {
        if(Timeout-- == 0)
        {
        /* Time-out occurred. Set LED to blinking mode */
        LED_Blinking(LED_BLINK_ERROR);
        }
      }
    #endif /* USE_TIMEOUT */
    }
    /* Clear flag ADC channel configuration ready */
    LL_ADC_ClearFlag_CCRDY(ADC1);
```
   7. /* Set ADC channels sampling time */
       LL_ADC_SetChannelSamplingTime(ADC1, LL_ADC_CHANNEL_VBAT, LL_ADC_SAMPLINGTIME_COMMON_1);

ycsin, 2022-03-15:
Is it possible to implement this directly in the zephyr's STM32 ADC driver?

FRASTM, 2022-03-15:
> Is it possible to implement this directly in the zephyr's STM32 ADC driver?

Trying, right now, but I'm not successful yet. I am afraid there are several issues, like writing the ADC register with ADEN or ADSTART <>0. 
FOR stm32G0 only (to be extended to other mcus) You could have a look at https://github.com/FRASTM/zephyr/tree/adc_g0 
```
ADC reading: 455 = 366 mV  
ADC reading: 453 = 364 mV  
ADC reading: 455 = 366 mV  
```

affadli, 2022-03-16:
> > Hi @FRASTM , I also got similar issue once using VBAT ADC but I used STM32WB chip. The ADC value that I got didnt match with battery.
> > Is there any proper example regarding ADC using VBAT channel?
> 
> With a simple stm32Cube application : [ADC_MultiChannelSingleConversion](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/NUCLEO-G071RB/Examples_LL/ADC/ADC_MultiChannelSingleConversion) modified to get internal VBAT on channel 14 in the **main.c**
> 
> 1. add a fourth element in the array to get the Vbat :  `#define ADC_CONVERTED_DATA_BUFFER_SIZE   (   4U)`
> 2. `__IO uint16_t uhADCxConvertedData_VBat_mVolt = 0U; `
> 3. `uhADCxConvertedData_VBat_mVolt = __LL_ADC_CALC_VREFANALOG_VOLTAGE(aADCxConvertedData[3], LL_ADC_RESOLUTION_12B);`
> 4. /* Set ADC measurement path to internal channels */
>    `LL_ADC_SetCommonPathInternalCh(__LL_ADC_COMMON_INSTANCE(ADC1), (LL_ADC_PATH_INTERNAL_VREFINT | LL_ADC_PATH_INTERNAL_TEMPSENSOR | LL_ADC_PATH_INTERNAL_VBAT));`
> 5. in the Configure_ADC
>    /* Set ADC group regular sequencer length and scan direction */
>    `LL_ADC_REG_SetSequencerLength(ADC1, LL_ADC_REG_SEQ_SCAN_ENABLE_4RANKS);`
> 6. /* Set ADC group regular sequence: channel on the selected sequence rank. */
> 
> ```
> LL_ADC_REG_SetSequencerRanks(ADC1, LL_ADC_REG_RANK_4, LL_ADC_CHANNEL_VBAT);
>    /* Poll for ADC channel configuration ready */
>    #if (USE_TIMEOUT == 1)
>    Timeout = ADC_CHANNEL_CONF_RDY_TIMEOUT_MS;
>    #endif /* USE_TIMEOUT */
>    
>    while (LL_ADC_IsActiveFlag_CCRDY(ADC1) == 0)
>    {
>    #if (USE_TIMEOUT == 1)
>      /* Check Systick counter flag to decrement the time-out value */
>      if (LL_SYSTICK_IsActiveCounterFlag())
>      {
>        if(Timeout-- == 0)
>        {
>        /* Time-out occurred. Set LED to blinking mode */
>        LED_Blinking(LED_BLINK_ERROR);
>        }
>      }
>    #endif /* USE_TIMEOUT */
>    }
>    /* Clear flag ADC channel configuration ready */
>    LL_ADC_ClearFlag_CCRDY(ADC1);
> ```
> 
> 7. /* Set ADC channels sampling time */
>    LL_ADC_SetChannelSamplingTime(ADC1, LL_ADC_CHANNEL_VBAT, LL_ADC_SAMPLINGTIME_COMMON_1);

Hi @FRASTM , I just wondering. So, in this case you mention that you're using channel14 to get vbat. Why not used VBAT channel?

ycsin, 2022-03-16:
> Hi @FRASTM , I just wondering. So, in this case you mention that you're using channel14 to get vbat. Why not used VBAT channel?

For G0B1, channel 14 is the VBAT channel

affadli, 2022-03-16:
> > Hi @FRASTM , I just wondering. So, in this case you mention that you're using channel14 to get vbat. Why not used VBAT channel?
> 
> For G0B1, channel 14 is the VBAT channel

I see, thats my mistakes. Thanks !

FRASTM, 2022-03-16:
yes, but it makes a difference in the adc driver when enabling channel 14 compared to enabling VBAT as the channel14 can also be for ADC (without any ref to VBAt ).

ycsin, 2022-03-18:
After going through #43863, I've tested my driver patch again and found that the VBAT can be read properly if `ADC_ACQ_TIME_MAX` is used as the acquisition time.

So, to read the VBAT:
1. Set the common path for the VBAT
2. Read the channel normally with `ADC_ACQ_TIME_MAX` as the acquisition time

closed 2022-07-01 by carlescufi