zephyrproject-rtos/zephyr #41775: Extend bt_le_scan_param.options to include channel filter

ryan4volts, 2022-01-13:
In order to minimize latency within my sleepy product, I need to optimise everything I can to get the smallest intervals between my peripheral waking up and my central receiving a packet from it. Security ended up not being a major thing, providing that the peripheral could not be tracked with standard private addressing and don't allow connections.

In doing so, I decided to go for an advertising scheme since a lvl 4 security repairing still takes some negotiations before it would be able to send alerts from the peierphal -> central.
Nevertheless,

My device is going to be in a field not really in close proximity to other devices, though I still want to stay away from flooding the network due to battery capacity constraints.

Reading around I found that I am able to advertise to a [single channel](https://github.com/zephyrproject-rtos/zephyr/pull/28759), but I couldn't find the functionality to scan a single (or reduced) channel set.

Currently I am scanning through all channels with short periods to pick up the single channel advertisement packets as quick as I can.

El-Mokhtar-AIssaoui, 2022-03-10:
@ryan4volts can you share the solution if you found it please?

ryan4volts, 2022-03-14:
None at this stage, have to work with the existing latency unless the feature gets added in the future.

hieuv, 2023-07-05:
I am not against adding channel filter to BLE scanning. However, for the purpose of minimizing the time from when the peripheral starts advertising to when the central detects it, I don't think doing advertising and scanning on a single channel will have a great effect.

When a device advertising on all three channels (37, 38, 39), it does so on all three channels, consecutively, within one advertising interval.

[image]
Reference image from [here](https://devzone.nordicsemi.com/f/nordic-q-a/66632/where-can-i-find-the-presentation-used-in-the-everything-you-need-to-know-about-bluetooth-low-energy-advertising-webinar).

Therefore, by limiting advertising to only one channel, in the best-case scenario, you only save the time to transmit two advertisement packets. That's not a lot.

kruithofa, 2023-07-06:
@ryan4volts @hieuv This is a proprietrary solution, not all controller implementation will necessarily implement this.

We will not work on this, but feel free to contribute by submitting a PR with an explanation on the improvement

closed 2023-07-06 by kruithofa