zephyrproject-rtos/zephyr #41776: LLVM: build failed with -fuse-ld=lld linker on qemu_x86.

chen-png, 2022-01-13:
**Describe the bug**
when tried to **use LLVM built-in lld linker to build zephyr case**, it met a **link error "section 'boot.text' will not fit in region 'RAM': overflowed by 18446744072631615488 bytes"**

**To Reproduce**
Steps to reproduce the behavior:
1. set LLVM toolchain:
     export ZEPHYR_TOOLCHAIN_VARIANT=llvm
     export LLVM_TOOLCHAIN_PATH=~/llvm/llvm-project/build_2
     (notice: **[](the LLVM version must be after this commit https://github.com/llvm/llvm-project/commit/2bf06d9345caeb26520be8e830c092683bbdf0f7, otherwise, it couldn't support parsing quoted symbol like "__start" which is used for zephyr entry point)**)
2. apply this patch

    
3.  west build -b qemu_x86 -p auto samples/hello_world/ -DCONFIG_LLVM_USE_BUILTIN_LLD=y
4. See error

**Logs and console output**
[83/97] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/chenpeng/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map
: && ccache **/home/chenpeng/llvm/llvm-project/build_2/bin/clang --target=i686-pc-none-elf**   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/x86/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/chenpeng/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/x86/core/libarch__x86__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/usr/lib64/gcc/x86_64-redhat-linux/9/32"  -L/home/chenpeng/zephyrproject/zephyr/build/zephyr  -lgcc  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  **-fuse-ld=lld**  -Wl,--orphan-handling=warn && cd /home/chenpeng/zephyrproject/zephyr/build/zephyr && /home/chenpeng/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E echo
clang-13: warning: argument unused during compilation: '-no-pie' [-Wunused-command-line-argument]
ld.lld: warning: <internal>:(.eh_frame) is being placed in '.eh_frame'
ld.lld: warning: <internal>:(.symtab) is being placed in '.symtab'
ld.lld: warning: <internal>:(.shstrtab) is being placed in '.shstrtab'
ld.lld: warning: <internal>:(.strtab) is being placed in '.strtab'
**ld.lld: error: no memory region specified for section '.eh_frame'
ld.lld: error: zephyr/linker_zephyr_pre0.cmd:101: unable to move location counter backward for: text
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705357312 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705357312 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705357473 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705357754 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358093 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358098 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358121 bytes**
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358134 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358149 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358191 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358239 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358523 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705358605 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705359498 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705359722 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705359870 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705359900 bytes
ld.lld: error: section 'text' will not fit in region 'RAM': overflowed by 18446744073705359914 bytes
ld.lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
collect2: error: ld returned 1 exit status
clang-13: error: linker (via gcc) command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/chenpeng/.local/bin/cmake --build /home/chenpeng/zephyrproject/zephyr/build

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: LLVM 
 - Zephyr Commit: 0e96f59e8365411dc9493

chen-png, 2022-01-14:
not sure why this "snippets-rom-start.ld" linker script for ROM is there, but for this case, it should does not matter.
because that linker script is used for putting .text section into ROM, but in this case, we put .text in RAM, so I removed it, then it didn't report any move backwards error or overflow error.
tested on qemu_x86, qemu_x86_64, built a hello_word and it worked well.

tejlmand, 2022-02-02:
> it's from a generated liner script "snippets-rom-start.ld", I am curious why we need to set the location counter to 0x0, because the RAM region it starts from 0x100000, this will meet a location counter move backwards error.

This originates from here, and is default included in the `snippets-rom-start.ld`:
https://github.com/zephyrproject-rtos/zephyr/blob/875ec0a0cb02d261f758adf2d0f6a01ca566b4df/arch/common/rom_start_offset.ld#L7-L8

but the x86 doesn't place its text in rom, as it don't have a FLASH memory region.
Seems `lld` is a bit more pedantic on this than `ld` on this.

One trick, is to set the rom offset == 0x100000, like this:
```
-DCONFIG_ROM_START_OFFSET=0x100000
```

This will fix both the overflow error and the location counter moving backwards.
We should look into if this snippet make sense for x86: https://github.com/zephyrproject-rtos/zephyr/blob/875ec0a0cb02d261f758adf2d0f6a01ca566b4df/include/arch/x86/ia32/linker.ld#L94

it probably does, but then we should ensure it behaves properly.

PR #41825 (2022-06-05): LLVM: add built-in lld linker support for x86.
add support to use LLVM built-in lld linker to build zephyr
for x86 platforms, mainly include:

add a CONFIG_LLVM_USE_LD to select the default linker for LLVM, ld or lld.
remove unsupported --sort-common option for lld; 
add linker command to handle .symtab, .strtab and .shstrtab sections for lld, because they will still be treated as a warning even with --orphan-handling=warn;
discard .eh_frame section if without CONFIG_EXCEPTIONS.
keep those sections related to ISR(.text.irqstubs, .gnu.linkonce.t.exc*) in output section for lld, because they won [... cut]
closed 2022-06-05 by carlescufi