zephyrproject-rtos/zephyr #41777: periodic_adv periodic_sync lost data

jenifer-su, 2022-01-13:
Hello,
i've two 52840DK,Windows10 X64,
D:\Program\ncs_v1.8.0\zephyr\samples\bluetooth\periodic_adv\
D:\Program\ncs_v1.8.0\zephyr\samples\bluetooth\periodic_sync\
using the BLE Zephyr controller.
now it work well,but when i change the interval time to 1s,
periodic_sync lost data packet 4 and 7. often lost.
how to advertising/receive continuous data (one by one and there is no repetition)?
how to change data lenght to big than 94?
and next i will change the timer interval 20ms,Is that ok?
[image]

[image]

thanks
Best regards,

jenifer-su, 2022-01-19:
Hello,

I have found the problem because the data + 1 is modified before send complete(occasionally). The sent callback function needs to be enabled and the data can be modified after send complete. However, the sent callback function is not entered at present and the reason is being found. please view to 

[nRF Connect SDK1.8.0 periodic_adv no recv callback - Nordic Q&A - Nordic DevZone - Nordic DevZone (nordicsemi.com)](https://devzone.nordicsemi.com/f/nordic-q-a/83770/nrf-connect-sdk1-8-0-periodic_adv-no-recv-callback)

thanks
Best regards,

carlescufi, 2022-03-02:
What you want to achieve is not really possible, because there is no guarantee of delivery for extended advertising packets, it's all best effort and some packets might be missed by the scanner, since there are no retransmissions to ensure all data reaches the scanner. Also, the application would have to be in sync with the clocks/timing of the controller to make sure it fits all slots with different indices.

closed 2022-03-02 by carlescufi