zephyrproject-rtos/zephyr #41784: virtio device driver

povergoing, 2022-01-13:
**Is your feature request related to a problem? Please describe.**
I want to use virtio device (front-end), I think there is no virtio-core framework, and so no way to use it.
For example, using qemu boot zephyr and set virtio-xxx-device switch to qemu .etc.

**Describe the solution you'd like**
If I want to add a virtio device, can I introduce virtio-core(virtio queue .. ) logic into zephyr?

**Describe alternatives you've considered**
Or reuse the virtio-core logic in OpenAMP components. But I think it's kind of wired if the virtio device is nothing related to it?

**Additional context**

carlocaione, 2022-11-04:
Preliminary abandoned work at https://github.com/Huawei-Paris-Research-Center/zephyr/tree/virtio-net

povergoing, 2022-11-04:
Thanks! I am going to close this issue, as I found that WindRiver guys are also doing this based on OpenAMP, https://github.com/OpenAMP/open-amp/tree/virtio-exp

closed 2022-11-04 by povergoing