zephyrproject-rtos/zephyr #41787: Alignment issue on STM32

jchabod, 2022-01-13:
**Describe the bug**

- I'll provide the project configuration as attachment

I'm running an application using sockets, but I got my application crashing.
Using STLINK debugger on STM32H board, I figured out that the exception is raised in "tcp2.c" on line `memcpy(buf.key, unique_key, sizeof(buf.key));`

Disassembly shows that the LDR command is realized on non 4-bytes aligned object. Ccehinkg in the zephyr.map file confirms that `unique_key` is on odd address.

I could temporarily fix and make the application not crash forcing the alignment, but I can guess that this is not the right correction:
```
diff --git a/arch/arm/core/aarch32/cortex_m/CMakeLists.txt b/arch/arm/core/aarch32/cortex_m/CMakeLists.txt
index 4e3050abd5..79a4f1c440 100644
--- a/arch/arm/core/aarch32/cortex_m/CMakeLists.txt
+++ b/arch/arm/core/aarch32/cortex_m/CMakeLists.txt
@@ -14,6 +14,7 @@ zephyr_library_sources(
   vector_table.S
   )

+zephyr_cc_option(-mno-unaligned-access)
 zephyr_library_sources_ifdef(CONFIG_USERSPACE thread.c)
 zephyr_library_sources_ifdef(CONFIG_DEBUG_COREDUMP coredump.c)
 zephyr_library_sources_ifdef(CONFIG_THREAD_LOCAL_STORAGE __aeabi_read_tp.S)
```

I could not find any CONFIG_xxx item that would solve that issue. (CORTEX seems to be able to support unaligned LDR using `SCB.CCR.UNALIGN_TRP`, is that configurable?)

**To Reproduce**
I could not reproduce the problem in a sample application because the arrays are always aligned in 4-bytes, however I declare variables.

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**
- ZEPHYR 2.7.0
- Build for BOARD stm32h735g_disco (but reproduceable on mimxrt1064_evk)
- host is LINUX WSL2 on W10.

**Additional context**

```
# General config
CONFIG_NEWLIB_LIBC=y
CONFIG_MAIN_STACK_SIZE=8192

CONFIG_SOCKS=y
CONFIG_POSIX_MAX_FDS=16
CONFIG_INIT_STACKS=y

CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV4_AUTO=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=5

CONFIG_NET_L2_DUMMY=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_LOOPBACK=y
CONFIG_NET_ARP=y
CONFIG_ETH_STM32_HAL=y
CONFIG_NET_MGMT=y
CONFIG_DNS_RESOLVER=y

CONFIG_NET_DEFAULT_IF_ETHERNET=y
CONFIG_NET_MAX_CONN=16
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_TX_STACK_SIZE=1500

CONFIG_NET_IF_MAX_IPV4_COUNT=2
CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=16
CONFIG_NET_BUF_DATA_SIZE=512

# Network address config
CONFIG_NET_DRIVERS=y
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_PROMISCUOUS_MODE=y
CONFIG_NET_CONFIG_INIT_TIMEOUT=3

CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8:100::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8:100::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.10.25"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.10.30"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.10.1"

CONFIG_PUBLISHER_ADDRESS="232.1.2.100:4840"
CONFIG_SUBSCRIBER_ADDRESS="232.1.2.101:4840"

CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y

# MbedTLS
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_DTLS=y
CONFIG_MBEDTLS_GENPRIME_ENABLED=y
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_HMAC_DRBG_ENABLED=y
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED=y
CONFIG_MBEDTLS_PEM_CERTIFICATE_FORMAT=y
CONFIG_MBEDTLS_SSL_EXPORT_KEYS=y
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=2048
CONFIG_MBEDTLS_TLS_VERSION_1_0=y
CONFIG_MBEDTLS_TLS_VERSION_1_1=y

CONFIG_MBEDTLS_USER_CONFIG_ENABLE=y
CONFIG_MBEDTLS_USER_CONFIG_FILE="config_custom_mbedtls.h"

# S2OPC
CONFIG_S2OPC=y

CONFIG_NET_SHELL=y

# for debug:
CONFIG_FAULT_DUMP=2

#for console:
CONFIG_CONSOLE_SUBSYS=y
CONFIG_CONSOLE_GETLINE=y
```

rlubos, 2022-01-21:
> Using STLINK debugger on STM32H board, I figured out that the exception is raised in "tcp2.c" on line memcpy(buf.key, unique_key, sizeof(buf.key));

> Disassembly shows that the LDR command is realized on non 4-bytes aligned object. Checking in the zephyr.map file confirms that unique_key is on odd address.

Both `buf.key`and `unique_key` are `uint8_t` arrays, I don't think the compiler should make any assumptions regarding alignment here, it seems more like a very low-level problem, not a networking one.

@carlescufi Who could look into the issue from the toolchain/architecture perspective, I don't think I'm the right assignee here.

jchabod, 2022-01-24:
> it seems more like a very low-level problem, not a networking one.

That's also what I'm thinking. I tried to explicitely force this alignment (patched the tcp2.c file adding attributes to the variables), But the software did not go really further (however, I did not try to identify the issue exact location because I could not find how to retrieve the failing address once in the exception vector, which make investigation... really slow

If you can help me with a procedure which gives me the addres of the exception once in `fault_s.S`, I sure could help.

rlubos, 2022-01-24:
> If you can help me with a procedure which gives me the addres of the exception once in `fault_s.S`, I sure could help.

The fault report that Zephyr produces on exception should contain the failing instruction address (look for `Faulting instruction address (r15/pc):`) and the thread that caused the crash, are you seeing this? If not, perhaps you need to enable the logger module.

carlescufi, 2022-01-26:
@jchabod at this point we really need more information to find out what is causing this. Please try to find a way to reproduce this reliably so we can then debug this, including boosting stack sizes in case there's a stack overflow that manifests itself as an unaligned access.

jchabod, 2022-01-26:
@rlubos => Ok, I added the configuration fr debug, And I now have that information.
@carlescufi  =>Yes I'll try that. (But I clearly have no stack issue, the .MAP file show the array on odd address, so the issue is at buld-time)

If you ahave any idea of any CONFG_xxx that could lead me to anything, do not hesitate to point me that out !

microbuilder, 2022-04-01:
> @rlubos => Ok, I added the configuration fr debug, And I now have that information.

Can you please share it here?

jchabod, 2022-04-05:
Hello, 
Sorry for very long delay, I did not have time to switch back on that subject...

I think that providing this information is not relevant here:
- The .map file sends an array on an odd address
- The memcpy is called on that array with optimized 32 bits access

=> This is not a run-time issue. This is a build-time inconsistency (which I temporarily solved by preventing optimized memcpy).

Note that I already tried on smaller reproducer, but of course it does not happen anymore...

I might soon be able to provide you a link to the full code

jchabod, 2022-04-05:
(For additionnal information, when I made the test forcing an even address, the soft crashed for the same reason on another unaligned address. I looked into the MPA file and found a large amount of unaligned array variables, that's why I did not go further)

jchabod, 2022-05-02:
Sorry, I have not been able to reproduce this issue. I now removed my patch and do not have anymore the problem.

I'll reopen, if this issue comes back in the future.

closed 2022-05-02 by jchabod