zephyrproject-rtos/zephyr #41792: CPU load halfed after PR #40784

alexsven, 2022-01-13:
**Describe the bug**
I noticed that the idle thread in simple samples like blinky is showing 49% cpu load. When checking out the commit just before the PR it shows 99% (as it should).
(https://github.com/zephyrproject-rtos/zephyr/pull/40784)

I've tested blinky for both nRF52840 and nRF5340, both show 49% CPU load for idle after PR, and 99% before the PR.

This PR was submitted by @peter-mitsis

peter-mitsis, 2022-01-13:
Investigating ....

peter-mitsis, 2022-01-13:
Alex,
What steps are you using to reproduce the problem? I ask because I am currently having difficulties just trying to build the blinky sample for the specified boards.

I've been trying to use ...
    west build -p -b <board name> ./samples/basic/blinky
... to just get a basic build of the sample project but it is giving me Kconfig build errors, for every board name that contains nrf52840 that I have tried so far.  At this point, I am assuming that either I have been picking the wrong board, and/or I have a knowledge gap.

nashif, 2022-01-13:
`west build -b nrf52840dk_nrf52811  ./samples/basic/blinky` works fine...

alexsven, 2022-01-14:
I used 
`west build -b nrf52840dk_nrf52840`
and 
`west build -b nrf5340dk_nrf5340_cpuapp`
depending on the board of course

nordic-krch, 2022-01-14:
I am able to reproduce it on qemu_86 and qemu_86_64.
For reference, commit before the break: 0bd3d46cfb86021629bc505f8ebfa3d949a20437.
Application i used: https://github.com/zephyrproject-rtos/zephyr/commit/36c210e06b424d34b837160e98cc45e3e4d400fc

Prints 0.0% on 0bd3d46cfb86021629bc505f8ebfa3d949a20437 and
```
Hello World! qemu_x86_64
load: 57.4%
load: 60.1%
load: 61.6%
load: 62.6%
load: 100.0%
```
on lastest.

peter-mitsis, 2022-01-14:
I'm able to duplicate it now--also on qemu_x86.

peter-mitsis, 2022-01-17:
I found at least one issue related to the updating of the system's thread_runtime_stats.
   - The total cycles (non-idle) is including the number of idle cycles.

Consequently, the system's execution cycles (sum of total_cycles and idle cycles) ultimately includes the idle cycles twice.

PR #41914 (2022-01-20): Update CPU runtime stats of non-idle time
Updates sched_cpu_update_usage() such that the CPU runtime stats
only update the its non-idle time when the current thread is not the
idle thread. This is necessary as otherwise the CPUs idle-time will
be double counted in k_thread_runtime_stats.execution_cycles.

Fixes: #41792
closed 2022-01-20 by nashif