zephyrproject-rtos/zephyr #41794: Bluetooth: ATT calls GATT callbacks with NULL conn pointer during disconnect

Thalley, 2022-01-13:
**Describe the bug**
When disconnected `att_reset` will be called, which will run through all outstanding requests and will call their respective callbacks. The issue is that the callback functions are called with `conn == NULL`: 

```
		if (req->func) {
			req->func(NULL, BT_ATT_ERR_UNLIKELY, NULL, 0,
				  req->user_data);
		}
```

This `NULL` reference is then propagated up to the application-defined callbacks. 

The GATT API does not state that the `conn` pointer in the callbacks can ever be NULL, and in almost all (if not all) cases the `conn` pointer is never NULL checked in callbacks for this reason. 

**To Reproduce**
It is timing related, as a disconnect need to happen after sending an ATT request, but before getting a response. Can be trigger by continuously sending any ATT/GATT request. 

**Expected behavior**
The callback should be called with the correct `conn` pointer, in order for the application to properly handle it. 

**Impact**
It is a fatal error in some cases (e.g. in VCS (`vcs.c`) where we need to the connection pointer to get the specific VCS instance), but otherwise will be application specific how bit an issue this is. 

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): N/A
 - Toolchain (e.g Zephyr SDK, ...): N/A
 - Commit SHA or Version used: 727ea490292255e3d0462f583fbc490a5c4936e4

**Additional context**
An alternative to fixing the pointer would be to document that the `conn` pointer can in fact be NULL in the `gatt.h` API.

Thalley, 2022-01-13:
@jhedberg @Vudentz I'd like your input on this, especially if it intentional that the callbacks are called with `conn == NULL`. Prior to calling the `req->func` the `att->conn` is also set to NULL.

jhedberg, 2022-01-13:
Have you been able to trace down at which point in the commit history this behavior was introduced? (eg using `git blame`). The commit messages might include some hints for the rationale (if one exists).

Thalley, 2022-01-14:
> Have you been able to trace down at which point in the commit history this behavior was introduced? (eg using `git blame`). The commit messages might include some hints for the rationale (if one exists).

The `req->func` has been called with `NULL` for the connection pointer ever since it was implemented 5 years ago. I'm amazed that we haven't seen more issues with this. 

The `att->conn = NULL` part was implemented as part of EATT 2 years ago, but since `att->conn` was not sent to the callback, it doesn't really matter.

jhedberg, 2022-02-02:
I suspect the original reasoning might have been to be able to distinguish between an actually received `ATT_UNLIKELY` error code over the air vs the connection going away (in which case you get a NULL pointer). @Vudentz can you confirm? If this reasoning makes sense, then maybe we should just update the documentation to state this special case?

jhedberg, 2022-02-02:
> It is a fatal error in some cases (e.g. in VCS (vcs.c) where we need to the connection pointer to get the specific VCS instance), but otherwise will be application specific how bit an issue this is.

That's indeed something that needs some solution. I suppose one option would be to include a connection reference as part of the callback's user data, however since the callback itself gets a bt_conn pointer, that doesn't strike me as a particularly clean solution.

Thalley, 2022-02-02:
> I suspect the original reasoning might have been to be able to distinguish between an actually received `ATT_UNLIKELY` error code over the air vs the connection going away (in which case you get a [...]

That's a good point, if a application cannot distinguish between a `ATT_UNLIKELY` from the remote device, or from a disconnect. We could possibly expose a `bt_conn_is_connected()` function that the application can call to get the connection state from the stack, and then determine if it is from a disconnect or not.

alwa-nordic, 2022-02-02:
> 

I agree. It's unfortunate that we alias `ATT_UNLIKELY` responses and failures further down the protocol stack. The problem is not only here. We are generating this synthetic `ATT_UNLIKELY` response several other places in the host, I think.

Vudentz, 2022-02-02:
> That's a good point, if a application cannot distinguish between a `ATT_UNLIKELY` from the remote device, or from a disconnect. We could possibly expose a `bt_conn_is_connected()` function that the [...]

I don't recall exactly why I used NULL, perhaps it was to prevent the application from reusing the conn since in most cases we should be checking if !conn but it doesn't seems to be the case for bt_conn_index.

Vudentz, 2022-02-02:
> > 
> 
> I agree. It's unfortunate that we alias `ATT_UNLIKELY` responses and failures further down the protocol stack. The problem is not only here. We are generating this synthetic `ATT_UNLIKELY` response several other places in the host, I think.

We could use zephyr specific errors if that really helps, but applications would need to understand them either way.

commit d54ae955 (2022-02-07): Bluetooth: ATT: Fix conn parameter to req cb in att_reset

The conn pointer is still valid / not reused at this time and can be
used further up the stack as an identifer. This simplifies the API of
ATT, and fixes callbacks in GATT that pass on this value directly since
their API does not allow conn to be NULL.

Fixes #41794
closed 2022-02-07 by carlescufi