zephyrproject-rtos/zephyr #41804: Include LVGL's MIT-licensed Kconfig in lib/gui/lvgl/

brgl, 2022-01-13:
## Origin

https://github.com/lvgl/lvgl/

## Purpose

This is the graphics library used in zephyr. It's currently at version 7 and https://github.com/zephyrproject-rtos/zephyr/pull/41336 is proposing its update to v8.

## Mode of integration

Only the Kconfig needs to be included in Zephyr.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/41336

## Description

Due to the problem described [here](https://github.com/zephyrproject-rtos/zephyr/pull/41336#issuecomment-1006887270), we need to include the upstream Kconfig locally until we rework how modules are handled in zephyr.

The upstream Kconfig needs to be included from lib Kconfigs but it will not be there if the LVGL module is not present.

# Dependencies

Zephyr will still depend on zephyr's LVGL fork.

## Revision

Actually the Kconfig from upstream LVGL needed to be adjusted for Zephyr's coding style so there's no upstream SHA yet. Once the updated version is upstream, I will send those changes to LVGL.

## License

MIT

kisvegabor, 2022-01-13:
Copied from [here](https://github.com/zephyrproject-rtos/zephyr/pull/41336#issuecomment-1012264443):

It's a little bit hard for me to foresee the legal consequences of having a dual-licensed file in an "plain" MIT repo.
AFAIK, dual licensing means people can choose the license which fits better. Can you confirm it? If so, I see no problem, as one of the licenses is still MIT.
@embeddedt can you comment on this?

brgl, 2022-01-14:
@kisvegabor I opened this in order to get the MIT licensed file into zephyr and spare you the headache. :)

kisvegabor, 2022-01-15:
Oh, cool :slightly_smiling_face:

brgl, 2022-01-20:
> After discussing this in the TSC, we want to try and do this the way SoF does it, by directly including the Kconfig in the module instead of copying it into the Zephyr tree. This has indeed some issues that might need solving, so I will ask @tejlmand to take a look together with me and see what we can do.

@carlescufi I thought we did agree that this is the right way to go but also that we'll merge the PR as is and then figure out how to put the Kconfig back into the lvgl repo (and do the same for other modules) after that in order not to block a significant feature?

brgl, 2022-01-20:
@carlescufi I clarified what needs doing during the dev-review meeting. I'm closing this as it's no longer needed.

closed 2022-01-20 by brgl