zephyrproject-rtos/zephyr #41816: nrf_802154 radio driver takes random numbers directly from entropy source

hubertmis, 2022-01-14:
It looks that the 802.15.4 driver for nrf platforms takes random number directly from system entropy source:
https://github.com/zephyrproject-rtos/zephyr/blob/449c37808a68824faa93b77fe24f39f05e9a9939/modules/hal_nordic/nrf_802154/radio/platform/nrf_802154_random_zephyr.c#L30

I think it should use Random Number Generation subsystem instead.
Is there any reason to use entropy source directly?

hubertmis, 2022-01-14:
@czeslawmakarski , @rlubos , do you know?

rlubos, 2022-01-14:
The code snippet you've posted only uses the entropy to seed the driver's internal software RNG, I don't think that's wrong? It'd not used for the actual random number generation.

hubertmis, 2022-01-14:
I think that's an acceptable, but probably unnecessary exhaustion of entropy source. I'm OK to keep it, but if there is no real reason I think it would be better to use `sys_rand32_get()` instead.

dleach02, 2022-01-18:
@hubertmis we are going to close this bug report. If you want to pursue this some more I recommend opening an enhancement request ticket. As pointed out, this function's usage of entropy is to seed a rng function. That is generally an okay usage of entropy although as you point out, it could have used sys_rand32_get() as well.

closed 2022-01-18 by dleach02