zephyrproject-rtos/zephyr #41818: In uart.h uart_irq_rx_ready() function not working properly for STM32F429 controller

SadanandPenore-eaton, 2022-01-14:
**Describe the bug**
In uart.h, Read_Int_status() function not working properly for STM32F429 controller.

When the byte is received on the Rx Line, the interupt is called.
At the start we need to check what type of interrupt is this. ( rx or tx ).
So for checking receive interrupt I put a check:
if( uart_irq_rx_ready() == 1 )

The device gives garbage behaviour.
The code can read the data if data is read without checking the uart_irq_rx_ready() function return.
But we need that function to know whether the interrupt is for receive or transmitt.

Same code is working for STM32F767ZI.
Appropriate changes are done in dts file.

Target --> STM32F429ZI
Zephyr Version ---> 2.7

Here is the code:

#include "Prod_Spec_UART.h"
#include "device.h"
#include <drivers/uart.h>
#include <stdio.h>

#define UART_2_DEVICE_NAME DT_LABEL(DT_NODELABEL(usart2))

int Read_Int_status(void );

const struct device* uart2_device = nullptr;
void Uart_Isr( const struct device* unused, void* user_data )
{
	unsigned char data;

	printf ("Byte Received\n" );// Printed on USB debug port
	uart_fifo_read( uart2_device, &data, 1U );	// Size = 1		
	printf ("Received Data is %d\n",data );

	if( Read_Int_status() == 1 )  // if this if is commented then only the receive is working properly.
	{
			uart_fifo_read( uart2_device, &data, 1U );	// Size = 1		
			printf ("Received Data is %d\n",data );
	}
}

void main( void )
{
	volatile unsigned int test_var = 0;
	struct uart_config m_uart_cfg;
	int dummy;
	uart2_device = device_get_binding(UART_2_DEVICE_NAME );

	m_uart_cfg.baudrate = 115200;
	m_uart_cfg.flow_ctrl = UART_CFG_FLOW_CTRL_NONE;
	m_uart_cfg.parity = UART_CFG_PARITY_NONE;
	m_uart_cfg.data_bits = UART_CFG_DATA_BITS_8;
	m_uart_cfg.stop_bits = UART_CFG_STOP_BITS_1;

	if ( uart2_device != nullptr )
	{
		uart_configure( uart2_device, &m_uart_cfg );
	}

	uart_irq_callback_set( uart2_device, &Uart_Isr );
	uart_irq_rx_enable( uart2_device );
	
	while(1)
	{
		test_var++;
		//Do nothing
	}
}

int Read_Int_status(void)
{
	int ret_val;
	ret_val = uart_irq_rx_ready( uart2_device );
	return ( ret_val );
}

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2022-01-17:
@SadanandPenore-eaton you say "Appropriate changes are done in dts file", do you mean adding an overlay or an entry in the nucleo_f429zi.dts board with
```
&usart2 {
	pinctrl-0 = <&usart2_tx_pd5
		     &usart2_rx_pd6
		     &usart2_rts_pd4
		     &usart2_cts_pd3>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};
```
 I expect the `CONFIG_UART_ASYNC_API=y`  is set too, so you probably have to set the DMA for uart transfer

SadanandPenore-eaton, 2022-01-18:
Oh. I found a problem area.
By the way I don't want to use DMA for now.
The problem I found was with uart_err_check(). -- (Sorry it is not pasted in the simplified code above.)

And here is the overlay file:
&usart2 {
	pinctrl-0 = <&usart2_tx_pd5 &usart2_rx_pd6>;
	current-speed = <115200>;
	status = "okay";
};

If I comment the function everything work fine.
This is simple code which receives the 4 bytes and transfers 4 bytes in return.
If uart_err_check() is called just before uart_irq_rx_ready(), the uart_irq_rx_ready() dosent return true.
refer the **bold** line of code below.

#include "device.h"
#include <drivers/uart.h>
#include <stdio.h>

#define UART_2_DEVICE_NAME DT_LABEL(DT_NODELABEL(usart2))
int Read_Int_status(void );
unsigned char rx_array[4];
int rx_index=0;
int tx_index=0;
const struct device* uart2_device = nullptr;
void Uart_Isr( const struct device* unused, void* user_data )
{
	unsigned char data;
	int error_status = 0;

	**//error_status = uart_err_check ( uart2_device ); // --> if this line is enabled the uart_irq_rx_ready() dosen't return true 
                                                                                       // even if bytes are received**

	if( uart_irq_rx_ready( uart2_device ) == 1 )
	{
			uart_fifo_read( uart2_device, &data, 1U );	// Size = 1		
			//printf ("Received Data is %d\n",data );
			rx_array[rx_index++]=data;
	}

	if ( uart_irq_tx_ready( uart2_device ) == 1 )
	{
		if ( tx_index < 4 )
		{
			uart_fifo_fill( uart2_device, reinterpret_cast<const uint8_t*>( &rx_array[tx_index++] ), 1U );	// Size = 1
		}
		else
		{
			uart_irq_tx_disable( uart2_device );
		}
	}

}

void main( void )
{
	volatile unsigned int test_var = 0;
	struct uart_config m_uart_cfg;
	int dummy;
	uart2_device = device_get_binding(UART_2_DEVICE_NAME );

	m_uart_cfg.baudrate = 115200;
	m_uart_cfg.flow_ctrl = UART_CFG_FLOW_CTRL_NONE;
	m_uart_cfg.parity = UART_CFG_PARITY_NONE;
	m_uart_cfg.data_bits = UART_CFG_DATA_BITS_8;
	m_uart_cfg.stop_bits = UART_CFG_STOP_BITS_1;

	if ( uart2_device != nullptr )
	{
		uart_configure( uart2_device, &m_uart_cfg );
	}

	uart_irq_callback_set( uart2_device, &Uart_Isr );
	uart_irq_rx_enable( uart2_device );
	
	while(1)
	{
		test_var++;
		if ( rx_index == 4 )
		{
			rx_index = 0;
			tx_index = 0;
			uart_irq_tx_enable( uart2_device );
			uart_fifo_fill( uart2_device, reinterpret_cast<const uint8_t*>( &rx_array[tx_index++] ), 1U );	// Size = 1
		}
	}
}

int Read_Int_status(void)
{
	int ret_val;
	ret_val = uart_irq_rx_ready( uart2_device );
	return ( ret_val );
}

SadanandPenore-eaton, 2022-01-18:
I came across https://github.com/zephyrproject-rtos/zephyr/issues/41339.
And added code related to line break at mentioned location. That also did not solve the problem.

FRASTM, 2022-01-19:
the  uart_err_check  is mapped to the uart_stm32_err_check that mainly deals with reception.
Maybe clearing the Error Flags also change the RXNE so that LL_USART_IsActiveFlag_RXNE is 0 (no data Rx)
I do not see the recommended "software sequence" in the uart_stm32_err_check as the RM0090 (for stm32F429) mentions : "It is cleared by a software sequence (an read to the USART_SR register followed by a read to the USART_DR register)."

Can you check if and which error was actually raised and cleared ?

closed 2022-02-01 by nashif