zephyrproject-rtos/zephyr #41821: ESP32 mcuboot bootloader failed

sam131208, 2022-01-14:
**Describe the bug**
Bluetooth initial failed after enabling BOOTLOADER_MCUBOOT.

**To Reproduce**
Steps to reproduce the behavior:
1. flash mcuboot.  ( https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md)
2. west build -b esp32 samples/bluetooth/peripheral/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y
3. west flash --bin-file build/zephyr/zephyr.signed.confirmed.bin

**Logs and console output**
 ============ PuTTY log 2022.01.13 11:34:27 ============
ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4604
ho 0 tail 12 room 4
load:0x40085000,len:14964
load:0x4009d000,len:3624
entry 0x40085478
[esp32] [INF] *** Booting MCUboot build v1.7.0-421-g399720d ***
[esp32] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] DRAM segment: start=0x54e8, size=0x17e4, vaddr=0x3ffbdb60
[esp32] [INF] IRAM segment: start=0x6ccc, size=0x115d0, vaddr=0x40080000
[esp32] [INF] start=0x4009014c
*** Booting Zephyr OS build v2.7.99-3070-g7eb50bebe486 ***
Bluetooth init failed (err -45)
[00:00:01.119,000] bt_settings: settings_subsys_init failed (err -45)

**Environment (please complete the following information):**
 - OS: Ubuntu

WilliamRoyle, 2022-01-16:
A few context questions, might be useful:

1. Exactly what ESP32 module are you using
2. Are we sure that bluetooth is supported on ESP32? It was not as of October https://www.cnx-software.com/2021/10/31/esp32-zephyr-nuttx-rtos-support/
3. Have you tested this without MCUboot to verify that MCUboot is the problem?

sam131208, 2022-01-16:
> A few context questions, might be useful:
> 
> 1. Exactly what ESP32 module are you using
The module is ESP32-DevKitC, support wifi , ble.

> 2. Are we sure that bluetooth is supported on ESP32? It was not as of October https://www.cnx-software.com/2021/10/31/esp32-zephyr-nuttx-rtos-support/

 Bluetooth is supported on ESP32, i am sure.

> 3. Have you tested this without MCUboot to verify that MCUboot is the problem?

I has test Hello World with MCUBOOT, it is work fine.
and test the project esp32 samples/bluetooth/peripheral without  enable BOOTLOADER_MCUBOOT, bluetooth is work fine.

sylvioalves, 2022-01-19:
@sam131208, the 3rd step is wrong. You should not flash `zephyr.signed.confirmed.bin`, but `zephyr.signed.bin`.  Besides this, I recommend you to run only `west flash`. As MCUBOOT is enabled, `west flash` will take care of flashing `zephyr.signed.bin` automatically. Would you try again?

sam131208, 2022-01-19:
@sylvioalves
 It seems can't bootload success.

Steps to reproduce the behavior:
   1.  flash mcuboot. ( https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md)
   2.  west build -b esp32 samples/bluetooth/peripheral/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y
   3.  west flash
        or   west flash --bin-file build/zephyr/zephyr.signed.bin

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7088
load:0x40078000,len:14312
load:0x40080400,len:3688
entry 0x40080678
I (52) boot: ESP-IDF release/v4.3-116-gbd2e2d8dc 2nd stage bootloader
I (52) boot: compile time 10:37:02
I (54) boot: chip revision: 1
I (58) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (74) boot.esp32: SPI Speed      : 40MHz
I (74) boot.esp32: SPI Mode       : DIO
I (74) boot.esp32: SPI Flash Size : 16MB
I (79) boot: Enabling RNG early entropy source...
I (84) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
I (95) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (110) boot:  2 factory          factory app      00 00 00010000 00100000
I (118) boot: End of partition table
E (122) esp_image: image at 0x10000 has invalid magic byte
W (128) esp_image: image at 0x10000 has invalid SPI mode 243
W (134) esp_image: image at 0x10000 has invalid SPI size 9
E (140) boot: Factory app partition is not bootable
E (146) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

sam131208, 2022-01-19:
@sylvioalves  It seems can't bootload success.

To Reproduce
Steps to reproduce the behavior:

    1. flash mcuboot. ( https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md)
         1) cmake -DCMAKE_TOOLCHAIN_FILE=tools/toolchain-esp32.cmake -DMCUBOOT_TARGET=esp32 -B build -GNinja
         2) cmake --build build/
         3) esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 1MB -o build/mcuboot_esp32.bin build/mcuboot_esp32.elf
         4) esptool.py -p /dev/ttyUSB0 -b 921600 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size 1MB --flash_freq 40m 0x00001000 build/mcuboot_esp32.bin
    2. west build -b esp32 samples/bluetooth/peripheral/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y
    3.  west flash

  Logs and console output:
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.01.19 16:20:59 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4604
ho 0 tail 12 room 4
load:0x40085000,len:14964
load:0x4009d000,len:3624
entry 0x40085478
[esp32] [INF] *** Booting MCUboot build v1.7.0-421-g399720d ***
[esp32] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] DRAM segment: start=0x54e8, size=0x17e4, vaddr=0x3ffbdb60
[esp32] [INF] IRAM segment: start=0x6ccc, size=0x115d0, vaddr=0x40080000
[esp32] [INF] start=0x4009014c
*** Booting Zephyr OS build v2.7.99-3070-g7eb50bebe486  ***
Bluetooth init failed (err -45)
[00:00:01.119,000] <err> bt_settings: settings_subsys_init failed (err -45)

sylvioalves, 2022-01-19:
@sam131208 I read you are using ESP32 DevkitC. It should have 4MB of flash. It means that `--flash_size 1MB` in both esptool.py commands should be replaced by `--flash_size 4MB`. Can you confirm that?

sam131208, 2022-01-20:
@sylvioalves  Thanks! you are right, the flash_size is flash chip size, not the image size. Now bootloader work fine.

closed 2022-01-20 by sam131208