zephyrproject-rtos/zephyr #41822: BLE IPSP sample cannot handle large ICMPv6 Echo Request

rlubos, 2022-01-14:
I've encountered a problem in the Bluetooth IPSP sample, the communication stalls after receiving large ICMPv6 Echo Request (>800 bytes of payload in my case). I think however that the issue is not related to the ping itself, but rather to large packet transmission in general. 

I can see in the logs that the ping request is processed successfully by the networking stack, but after passing the reply to the IPSP L2 layer, the response is not sent and the TX side becomes unresponsive (no more replies are sent, even smaller ones):
```
/* Smaller ping received and replied with success */
[00:04:40.069,915] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:40.073,791] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:40.075,622] <dbg> net_bt.ipsp_recv: (0x20001a40): Incoming data channel 0x20000170 len 246                                                                                                                    
[00:04:40.075,836] <dbg> net_bt.net_bt_recv: (0x20001e68): iface 0x200012d4 pkt 0x2000a144 len 246                                                                                                                   
[00:04:40.075,958] <dbg> net_icmpv6.net_icmpv6_input: (0x20001e68): ICMPv6 Echo Request received type 128 code 0                                                                                                     
[00:04:40.076,019] <dbg> net_icmpv6.icmpv6_handle_echo_request: (0x20001e68): Received Echo Request from 2001:db8::2 to 2001:db8::1                                                                                  
[00:04:40.076,263] <dbg> net_icmpv6.icmpv6_handle_echo_request: (0x20001e68): Sending Echo Reply from 2001:db8::1 to 2001:db8::2                                                                                     
[00:04:40.076,324] <dbg> net_bt.net_bt_send: (0x20001e68): iface 0x200012d4 pkt 0x2000a3ec len 248

/* Large ping gets stuck on the TX */                                                                                                                   
[00:04:49.169,647] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.173,522] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.175,445] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.179,321] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.183,197] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.187,042] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.190,917] <dbg> net_bt.ipsp_alloc_buf: (0x20001a40): Channel 0x20000170 requires buffer                                                                                                                     
[00:04:49.193,359] <dbg> net_bt.ipsp_recv: (0x20001a40): Incoming data channel 0x20000170 len 846                                                                                                                    
[00:04:49.193,542] <dbg> net_bt.net_bt_recv: (0x20001e68): iface 0x200012d4 pkt 0x2000a144 len 846                                                                                                                   
[00:04:49.193,756] <dbg> net_icmpv6.net_icmpv6_input: (0x20001e68): ICMPv6 Echo Request received type 128 code 0                                                                                                     
[00:04:49.193,817] <dbg> net_icmpv6.icmpv6_handle_echo_request: (0x20001e68): Received Echo Request from 2001:db8::2 to 2001:db8::1                                                                                  
[00:04:49.194,305] <dbg> net_icmpv6.icmpv6_handle_echo_request: (0x20001e68): Sending Echo Reply from 2001:db8::1 to 2001:db8::2                                                                                     
[00:04:49.194,396] <dbg> net_bt.net_bt_send: (0x20001e68): iface 0x200012d4 pkt 0x2000a3ec len 848                                                                                                                                                     
[00:04:49.194,702] <wrn> bt_l2cap: No credits to transmit packet
```

I'm not sure how severe `No credits to transmit packet` warning is, I've observed this warning occasionally with smaller packets as well, but it does not end up in a communication freeze.

**To Reproduce**
Build the sample according to instructions in https://docs.zephyrproject.org/2.6.0/samples/bluetooth/ipsp/README.html, and connect it to the host (tested with `nrf52840dk_nrf52840`). Observe that smaller pings are handled w/o issue:
> ping -6 -I bt0 2001:db8::1 -c 1

but the communication gets stuck after a larger one:

> ping -6 -I bt0 2001:db8::1 -s 800 -c 1

**Expected behavior**
Ideally, the sample should be able to reply to large pings, at very least it should still be responsive if it's not able to reply.

**Environment (please complete the following information):**
 - Manjaro Linux, kernel version 5.10.79-1
 - Zephyr SDK 0.13.1
 - Zephyr commit 449c37808a68824faa93b77fe24f39f05e9a9939
 - nrf52840dk_nrf52840

pizi-nordic, 2022-01-14:
Also note very high RAM usage of this sample - it barely fits in nRF52832...

rlubos, 2022-01-19:
No no, I've reported the issue because I'm not able to resolve it myself, I shouldn't be assigned to it.

cvinayak, 2022-04-27:
@rlubos If the solution is restricted due to RAM requirement, could you mark this as a known issue?

closed 2022-11-02 by alwa-nordic