zephyrproject-rtos/zephyr #41823: Bluetooth: Controller: llcp: Remote request are dropped due to lack of free proc_ctx

ppryga, 2022-01-14:
**Describe the bug**
The issues is related with refactored LLCPs implementation.

Two devices are in a connected state and start to exchange messages.
First device host starts `CONFIG_BT_CTLR_LLCP_PROC_CTX_BUF_NUM` number of control procedures by HCI commands.
Second device starts at leas one remote control procedure.

First device receives remote request and acknowledges it (default behavior of Controller).
New procedure is handled by upper link layer but due to lack of free `proc_ctx` the command is dropped without and error or notification.

**To Reproduce**
Use peripheral and central samples. Modify them to run multiple LLCPs on one end and at leas one LLCP on other end just after connection is established.

**Expected behavior**
Remote control procedure should be handled accordingly or not acknowledged.

**Impact**
Easy to reproduce with `CONFIG_BT_CTLR_LLCP_PROC_CTX_BUF_NUM` to default value.
Can cause strange and not expected behavior due to lack of notification about procedure drop.

kruithofa, 2022-01-20:
It would be good to have some (unit) tests for this behaviour

erbr-ot, 2022-06-30:
Reopening, as this issue is still present, however in slightly modified incarnation.
With a split of the local vs remote procedure contexts the dependency on local context has been removed, however there is still a limitation re. number of remote contexts across multiple connections. 
Thus with a number of remote CTXs less than number of connections, a connection could get starved resulting in 'loss of pdu' and consequently loss of connection. 

One way to handle is to queue up a PDU that 'can not get a CTX' and check back later - one kind of can'o'wormes. @thoh-ot has done a design for this.  
Another would be to find a way to NACK the PDU that 'can not get a CTX' this way pushing back to the peer - another ...

mmahadevan108, 2022-09-09:
@erbr-ot can you test again. Per Issue [ 15256](https://github.com/zephyrproject-rtos/zephyr/issues/15256), this should be fixed.

erbr-ot, 2022-09-12:
This is still an issue, as the handling of it was not yet implemented.

kruithofa, 2022-09-13:
I am working on this issue

kruithofa, 2022-09-13:
Short term solution: 
- Ensure that each connection has (at least) 1 procedure context available, and disconnect a connection when it initiates a 2nd procedure before the 1st is finished
Disadvantage: memory consumption

Long term solution:
- Create new API in LLL for NACK-ing control PDUs on a connection, requires check on LL-id and if it is a request for a new procedure
- Use a function-pointer for the normal case vs. case where no procedure-context is available
-- Function 1: function as is today
-- Function 2: check for LL-id, check if the incoming PDU is acceptable by the local procedure. If not: sent NACK
Disadvantage: added processing, but can be OK since that is only in the exception case
First step for the long term solution: create an MSC to show proof of concept

Action items:
- Create MSC to show  proof of concept

kruithofa, 2022-10-13:
PR #51239 is possibly blocked by this issue

closed 2023-01-02 by github-actions[bot]

ppryga-nordic, 2023-01-02:
@kruithofa was the issue fixed?