zephyrproject-rtos/zephyr #41826: MQTT connection failed

maweicoltsmart, 2022-01-14:
**Describe the bug**
branch：main
zephyr/samples/net/mqtt_publisher/

The demo program runs fine, but when I make the following modifications, the MQTT connection fails.

```
diff --git a/samples/net/mqtt_publisher/src/main.c b/samples/net/mqtt_publisher/src/main.c
index 7d7163aeab..ed3f01f806 100644
--- a/samples/net/mqtt_publisher/src/main.c
+++ b/samples/net/mqtt_publisher/src/main.c
@@ -303,8 +303,8 @@ static void client_init(struct mqtt_client *client)
        /* MQTT client configuration */
        client->broker = &broker;
        client->evt_cb = mqtt_evt_handler;
-       client->client_id.utf8 = (uint8_t *)MQTT_CLIENTID;
-       client->client_id.size = strlen(MQTT_CLIENTID);
+       client->client_id.utf8 = "2021121800000003";
+       client->client_id.size = 16;
        client->password = NULL;
        client->user_name = NULL;
        client->protocol_version = MQTT_VERSION_3_1_1;
```

log shows as follows：

```
[00:00:03.271,000] <inf> net_mqtt_publisher_sample: attempting to connect: 
[00:00:03.282,000] <dbg> net_mqtt_sock_tcp.mqtt_client_tcp_connect: (0x20003738): Created socket 5
[00:00:03.286,000] <dbg> net_core.net_recv_data: (0x200005d8): prio 0 iface 0x200015e8 pkt 0x2002cc70 len 64
[00:00:03.286,000] <dbg> net_core.net_rx: (0x20004c80): Received pkt 0x2002cc70 len 64
[00:00:03.286,000] <dbg> net_core.processing_data: (0x20004c80): Consumed pkt 0x2002cc70
[00:00:03.292,000] <dbg> net_core.net_recv_data: (0x200005d8): prio 0 iface 0x200015e8 pkt 0x2002cc70 len 64
[00:00:03.292,000] <dbg> net_core.net_rx: (0x20004c80): Received pkt 0x2002cc70 len 64
[00:00:03.293,000] <dbg> net_ipv4.net_ipv4_input: (0x20004c80): IPv4 packet received from 192.168.10.161 to 192.168.10.135
[00:00:03.293,000] <dbg> net_core.processing_data: (0x20004c80): Dropping pkt 0x2002cc70
[00:00:03.294,000] <dbg> net_mqtt_sock_tcp.mqtt_client_tcp_connect: (0x20003738): Connect completed
[00:00:03.294,000] <dbg> net_mqtt_enc: Encoding Protocol Description.
                                       4d 51 54 54                                      |MQTT             
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_utf8_str: (0x20003738): >> str_size:00000006 cur:0x20008a70, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint16: (0x20003738): >> val:0004 cur:0x20008a70, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.connect_request_encode: (0x20003738): Encoding Protocol Version 04.
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint8: (0x20003738): >> val:04 cur:0x20008a76, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint8: (0x20003738): >> val:00 cur:0x20008a77, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.connect_request_encode: (0x20003738): Encoding Keep Alive Time 003c.
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint16: (0x20003738): >> val:003c cur:0x20008a78, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc: Encoding Client Id.
                                       32 30 32 31 31 32 31 38  30 30 30 30 30 30 30 33 |20211218 00000003
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_utf8_str: (0x20003738): >> str_size:00000012 cur:0x20008a7a, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint16: (0x20003738): >> val:0010 cur:0x20008a7a, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.mqtt_encode_fixed_header: (0x20003738): << msg type:0x10 length:0x0000001c
[00:00:03.294,000] <dbg> net_mqtt_enc.packet_length_encode: (0x20003738): >> length:0x0000001c cur:(nil), end:(nil)
[00:00:03.294,000] <dbg> net_mqtt_enc.mqtt_encode_fixed_header: (0x20003738): Fixed header length = 02
[00:00:03.294,000] <dbg> net_mqtt_enc.pack_uint8: (0x20003738): >> val:10 cur:0x20008a6e, end:0x20008e6b
[00:00:03.294,000] <dbg> net_mqtt_enc.packet_length_encode: (0x20003738): >> length:0x0000001c cur:0x20008a6f, end:0x20008e6b
[00:00:03.296,000] <dbg> net_mqtt.client_connect: (0x20003738): Connect completed
[00:00:03.338,000] <dbg> net_core.net_recv_data: (0x200005d8): prio 0 iface 0x200015e8 pkt 0x2002cc70 len 64
[00:00:03.338,000] <dbg> net_core.net_rx: (0x20004c80): Received pkt 0x2002cc70 len 64
[00:00:03.338,000] <dbg> net_ipv4.net_ipv4_input: (0x20004c80): IPv4 packet received from 192.168.10.161 to 192.168.10.135
[00:00:03.338,000] <dbg> net_core.processing_data: (0x20004c80): Dropping pkt 0x2002cc70
[00:00:05.296,000] <dbg> net_mqtt_sock_tcp.mqtt_client_tcp_disconnect: (0x20003738): Closing socket 5
```

When I put a letter in front of the above modified clientid, MQTT can also connect fine, I don't know why.

```
diff --git a/samples/net/mqtt_publisher/src/main.c b/samples/net/mqtt_publisher/src/main.c
index 7d7163aeab..08f6bd7b97 100644
--- a/samples/net/mqtt_publisher/src/main.c
+++ b/samples/net/mqtt_publisher/src/main.c
@@ -303,8 +303,8 @@ static void client_init(struct mqtt_client *client)
        /* MQTT client configuration */
        client->broker = &broker;
        client->evt_cb = mqtt_evt_handler;
-       client->client_id.utf8 = (uint8_t *)MQTT_CLIENTID;
-       client->client_id.size = strlen(MQTT_CLIENTID);
+       client->client_id.utf8 = "M2021121800000003";
+       client->client_id.size = 17;
        client->password = NULL;
        client->user_name = NULL;
        client->protocol_version = MQTT_VERSION_3_1_1;

```

rlubos, 2022-01-14:
@maweicoltsmart I'm not able to reproduce the issue, using local `mosquitto` broker and the client ID as in your example it works just fine. Could you provide more details, i. e. what broker do you use, on what platform? A wireshark log of the failed connection attempt would also be helpful.

maweicoltsmart, 2022-01-14:
Sorry, this is my own mistake.

closed 2022-01-14 by maweicoltsmart