zephyrproject-rtos/zephyr #41830: CONF_FILE, OVERLAY_CONFIG parsing expands ${ZEPHYR_<whatever>_MODULE_DIR}

gregshue, 2022-01-14:
**Is your enhancement proposal related to a problem? Please describe.**
Downstream project integrators may be pulling content in from multiple downstream repositories, and may be configuring multiple similar projects. In order to reference config files in another module, integrators must explicitly provide either absolute pathnames, project-dir-relative pathnames, or ZEPHYR_BASE-prefixed pathnames to each file. This solution does not reflect that the mounting points of ZEPHYR_MODULES is arbitrary and is difficult to maintain when an imported west manifest changes the path to a module.

**Describe the solution you'd like**
I would like to see the parsing of the CMake variables CONF_FILE, OVERLAY_CONFIG expand ${ZEPHYR_MYMODULE_MODULE_DIR} substrings into the absolute path to the referenced module.  This enables integrators to reference overlay files in a different module without caring where that module is located in the workspace.

It seems this can be easily implemented where needed similar to the following example:
Replace: 
    `string(REPLACE " " ";" OVERLAY_CONFIG_AS_LIST "${OVERLAY_CONFIG}")`
With:
    `string(CONFIGURE "${OVERLAY_CONFIG}" OVERLAY_CONFIG_EXPANDED)`
    `string(REPLACE " " ";" OVERLAY_CONFIG_AS_LIST "${OVERLAY_CONFIG_EXPANDED}")`

As of Zephyr 2.7.0 it appears this replacement is only needed in `cmake/app/boilerplate.cmake` and `cmake/kconfig.cmake`.

**Describe alternatives you've considered**
I have not considered any other options.

**Additional context**

gregshue, 2022-02-11:
@tejlmand Should we be able to use "\${ZEPHYR_CURRENT_MODULE_DIR}" in the same way?

tejlmand, 2022-02-21:
no, cause that setting is empty when you try to expand it using `string(CONFIGURE ...)`.

Exactly showing why `string(CONFIGURE ...)` must be used with care, cause people often fail to understand how it works.

commit cfcf6301 (2022-10-13): build: Overlay build vars expand ${ZEPHYR_<module>_MODULE_DIR}

Support referencing module directories by name in CONF_FILE,
OVERLAY_CONFIG, and DTC_OVERLAY_FILE so that projects can reference
overlay files in arbitrary modules.

Verified by passing all the following tests:
  ./scripts/twister -T tests/cmake/overlays/

Fixes #41830
closed 2022-10-13 by mbolivar-nordic