zephyrproject-rtos/zephyr #41836: UART ESP32 - UART1 & UART2 Not Working

WilliamRoyle, 2022-01-15:
Hello, 

I am trying to get multiple UART connections running on the ESP32 (In particular, the Huzzah board). The devicetree for the ESP32 is as follows:

``

Meanwhile, my code is as follows:

`const struct device *uart_dev = DEVICE_DT_GET(DT_NODELABEL(uart0));`

Finally, I receive this error on building the project.

Any thoughts? Not sure why th

sylvioalves, 2022-01-19:
@WilliamRoyle, please, would you test this issue using this https://github.com/zephyrproject-rtos/zephyr/pull/41941?

LixinGuoX, 2022-01-19:
I have problem on building with this patch, is this a common issue?
```
/home/guo/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.bfd: zephyr/drivers/serial/libdrivers__serial.a(uart_esp32.c.obj):(.literal.uart_esp32_configure_pins+0x0): undefined reference to `uart_periph_signal'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/guo/zephyrproject/zephyr/build
```

sylvioalves, 2022-01-19:
@LixinGuoX, please, run `west update` so that `hal_espressif` brings that symbol.

sylvioalves, 2022-01-19:
@LixinGuoX Can we close this?

sylvioalves, 2022-01-21:
Closing. Re-open if needed.

closed 2022-01-21 by sylvioalves

WilliamRoyle, 2022-01-22:
Sylvio,

Thank you so much for your help with this. Unfortunately, I won't be able to test - long story Glad that this is cleared up in the code base.

PR #41941 (2022-01-21): driver: uart: esp32: fix uart macro and initialization
The main purpose of this PR is to fix a few uart issues:

Current uart driver implementation is uncompleted regarding the
usage of DT_INST_FOREACH_STATUS_OKAY. If uart0 and uart2 are selected,
build breaks due to peripheral number ordering, which would be
0 and 1 in this case. This fix PR fix this by re-working the macros
and setting proper uart peripheral instances in DTSI, required for signal
routing configuration.

Uart console cannot be installed due to its driver level initialization.
This was fixed be changing the level value.

Remov [... cut]