zephyrproject-rtos/zephyr #41861: tests: kernel: There are no log output after flashing image to ADSP of Up Xtreme

mengxianglinx, 2022-01-17:
**Describe the bug**
For some test cases, binary image can be downloaded to ADSP, but there is output can be caught by adsplog.py, so those test cases failed with "timeout"

 - Platform: ADSP on Up Xtreme

**To Reproduce**
Steps to reproduce the behavior:
1. cavsload.sh:
```
set -e

FLASHER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v25.py
LOGGER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
if [ "$1" = "" ]; then
        ssh root@localhost pkill -9 -f adsplog.py || true
        sleep 1
        ssh root@localhost "python3 -u $LOGGER --no-history || true"
        exit 0
fi

BLDDIR=$1

RIMAGE=$HOME/twister_tools/rimage
KEY=$HOME/twister_tools/keys/otc_private_key_3k.pem
IMAGE=$BLDDIR/zephyr/zephyr.ri
ELF=$BLDDIR/zephyr/main.mod
BOOT=$BLDDIR/zephyr/boot.mod

$RIMAGE -v -k $KEY -o $IMAGE -c $HOME/twister_tools/config/tgl.toml \
        -i 3 -e $BOOT $ELF > $BLDDIR/rimage.log

while [ ! -e $IMAGE ]; do
    sleep 0.1
done

ssh root@localhost pkill -9 -f cavs-fw-v25.py || true
sleep 1
ssh root@localhost "$FLASHER $IMAGE || true"
```
2. twister -p intel_adsp_cavs25 --device-testing --device-serial-pty=/path/to/cavsload.sh --west-flash=/path/to/cavsload.sh -T tests/ -j 1
3. The failed cases:
tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random                                                                                                                                                                tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap
tests/kernel/fatal/exception/kernel.common.stack_sentinel
tests/kernel/events/event_api/kernel.events
tests/kernel/threads/thread_stack/kernel.threads.thread_stack
tests/kernel/device/kernel.device.pm
tests/kernel/device/kernel.device
tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe
tests/kernel/obj_tracking/kernel.objects.tracking

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 44462723f5

dcpleung, 2022-01-19:
The `tests/kernel/events/event_api/kernel.events` is fixed by #41936.

Others except `tests/kernel/fata/exception/`, `tests/kernel/obj_tracking/` and `tests/kernel/events/event_api` are running fine on my Chromebook.

enjiamai, 2022-01-21:
Sorry, I think it should be medium bug because it is not a blocker for zephyr mainline

mengxianglinx, 2022-01-21:
normally twister call adsplog.py before cavs-fw-v25.py, if call adsplog.py immediately after cavs-fw-v25.py, with the modification described at https://github.com/zephyrproject-rtos/zephyr/issues/41862#issuecomment-1018509021, most of failed cases will pass

teburd, 2022-01-21:
This is fixed with #42022 as far as I can tell, please check

andyross, 2022-01-22:
Rename to use a clear platform name.  Though I did it wrong the first time :)

mengxianglinx, 2022-01-22:
with the latest cavstwist.sh, these test cases all passed. close it.

closed 2022-01-22 by mengxianglinx