zephyrproject-rtos/zephyr #41862: tests: kernel: fail to download firmware to ADSP on Up Xtreme

mengxianglinx, 2022-01-17:
**Describe the bug**
Fail to download binary image of test cases to ADSP on Up Xtreme, and the test cases failed with "timeout"

 - Platform: ADSP on Up Xtreme

**To Reproduce**
Steps to reproduce the behavior:
1. cavsload.sh:
```
set -e

FLASHER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v25.py
LOGGER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
if [ "$1" = "" ]; then
        ssh root@localhost pkill -9 -f adsplog.py || true
        sleep 1
        ssh root@localhost "python3 -u $LOGGER --no-history || true"
        exit 0
fi

BLDDIR=$1

RIMAGE=$HOME/twister_tools/rimage
KEY=$HOME/twister_tools/keys/otc_private_key_3k.pem
IMAGE=$BLDDIR/zephyr/zephyr.ri
ELF=$BLDDIR/zephyr/main.mod
BOOT=$BLDDIR/zephyr/boot.mod

$RIMAGE -v -k $KEY -o $IMAGE -c $HOME/twister_tools/config/tgl.toml \
        -i 3 -e $BOOT $ELF > $BLDDIR/rimage.log

while [ ! -e $IMAGE ]; do
    sleep 0.1
done

ssh root@localhost pkill -9 -f cavs-fw-v25.py || true
sleep 1
ssh root@localhost "$FLASHER $IMAGE || true"
```
2. twister -p intel_adsp_cavs25 --device-testing --device-serial-pty=/path/to/cavsload.sh --west-flash=/path/to/cavsload.sh -T tests/
3. in device.log:
 INFO:cavs-fw:cAVS firmware loader v25
 IFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
 INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
 INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
 INFO:cavs-fw:Detected cAVS 1.8+ hardware
 INFO:cavs-fw:Reset HDA device
 INFO:cavs-fw:Enable HDA global processing
 INFO:cavs-fw:Powering down, ADSPCS = 0x1010e0e
 INFO:cavs-fw:Powered down, ADSPCS = 0xf0f
 INFO:cavs-fw:Configuring DMA output stream 7...
 INFO:cavs-fw:Obtained the physical address of the mapped huge page.
 INFO:cavs-fw:Mapped 2M huge page at 0x15d200000 to contain 106496 bytes of firmware
 INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
 INFO:cavs-fw:Powering up DSP core #0, ADSPCS = 0xf0f
 INFO:cavs-fw:Powered up 1 cores, ADSPCS = 0x1010f0f
 INFO:cavs-fw:Sending PURGW_FW IPC, HIPCR = 0x81004000
 INFO:cavs-fw:Starting 1 cores, ADSPCS = 0x1010f0f
 INFO:cavs-fw:Started 1 cores, ADSPCS = 0x1010e0e
 INFO:cavs-fw:Waiting for ROM init, FW_STATUS = 0x5000001
 INFO:cavs-fw:ROM ready, FW_STATUS = 0x5000001
 INFO:cavs-fw:Awaiting IPC acknowledgment, HIPCA 0x80000000
 INFO:cavs-fw:Awaiting ROM init... FW_STATUS = 0x5000001
 INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
 INFO:cavs-fw:Waiting for load, FW_STATUS = 0x5000001
 **WARNING:cavs-fw:Load failed?  FW_STATUS = 0x80000006**
 INFO:cavs-fw:ADSPCS = 0x1010e0e
 INFO:cavs-fw:cAVS v25 firmware load complete, 1 cores active

4. Failed cases:
tests/kernel/context/kernel.common
tests/kernel/msgq/msgq_api/kernel.message_queue
tests/kernel/fatal/message_capture/kernel.logging.message_capture
tests/kernel/lifo/lifo_api/kernel.lifo
tests/kernel/fifo/fifo_usage/kernel.fifo.usage
tests/kernel/workq/work/kernel.work.api
tests/kernel/sched/deadline/kernel.scheduler.deadline
tests/kernel/sched/schedule_api/kernel.scheduler.multiq
tests/kernel/semaphore/semaphore/kernel.semaphore
tests/kernel/timer/timer_error_case/kernel.timer.error_case

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 44462723f5

andyross, 2022-01-18:
That's a load failure, which usually implies a glitch in the DSP (it's possible for the DSP to get into a state from which it can't recover, at least with our loader.  This is a known thing).  Are you absolutely sure that this is repeating after a power cycle of the device, and that it affects only those specific Zephyr tests and not just "every test you try after it hangs?".

It seems very unlikely that this is a Zephyr bug to be fixed, basically.  Can you retest?

mengxianglinx, 2022-01-19:
Every time run twister, about a dozen cases failed because of "WARNING:cavs-fw:Load failed? FW_STATUS = 0x80000006", but they are not always the same cases.

andyross, 2022-01-19:
Again, that's known behavior.  One test hangs the DSP and causes spurious failures on other tests.

The test rig needs to be robust enough to be able to retry tests after a reboot to figure out.  If there are tests that reliably hang the DSP (one was discovered yesterday) we need to know what they are.  It's not enough to just run twister and walk away. You need to help us isolate failing test cases.

andyross, 2022-01-19:
Here are some more specific suggestions:

1. Always inspect the logs after a failure like this to find the first case that failed.  Add that to a list of suspected "DSP hang tests"
2. Validate the test with several manual runs to make sure you can see it fail using west/cavs-fw*.py/adsplog.py.
3. Submit a github bug for that one test.  Mention that it doesn't just fail but hangs the device.
4. Filter that test from your rig when running twister in the future

mengxianglinx, 2022-01-20:
"One test hangs the DSP and causes spurious failures on other tests" has been reported in #41877, it is closed. Another case with the same symptom is reported to #41989.
Failed test cases listed here do not cause following cases failed or cause DSP hang.

mengxianglinx, 2022-01-21:
If download firmware failed, re-download it will most likely success. refine cavsload.sh like:
```
set -e

if [ "$1" != "" ]; then
        echo $1 > $ZEPHYR_BASE/inform_logger
        exit 0
fi

while [ ! -e $ZEPHYR_BASE/inform_logger ]; do
        sleep 1
done
BLDDIR=`(cat $ZEPHYR_BASE/inform_logger)`
rm $ZEPHYR_BASE/inform_logger

RIMAGE=$HOME/twister_tools/rimage
KEY=$HOME/twister_tools/keys/otc_private_key_3k.pem
IMAGE=$BLDDIR/zephyr/zephyr.ri
ELF=$BLDDIR/zephyr/main.mod
BOOT=$BLDDIR/zephyr/boot.mod

$RIMAGE -v -k $KEY -o $IMAGE -c $HOME/twister_tools/config/tgl.toml \
        -i 3 -e $BOOT $ELF > $BLDDIR/rimage.log

while [ ! -e $IMAGE ]; do
    sleep 0.1
done

FLASHER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v25.py
LOGGER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
ssh root@localhost pkill -9 -f cavs-fw-v25.py || true
ssh root@localhost pkill -9 -f adsplog.py || true
sleep 1

**ssh root@localhost "python3 $FLASHER $IMAGE || python3 $FLASHER $IMAGE || true"**
ssh root@localhost "python3 -u $LOGGER || true"
```
in cavs-fw-v25.py, add
```
 sd.CTL = 1
time.sleep(1)
exit(1)
```
under line
```
log.warning(f"Load failed?  FW_STATUS = 0x{dsp.SRAM_FW_STATUS:x}")"
```
these modification make the pass rate rise to 90%

andyross, 2022-01-21:
Going to close this.  We have multiple people reporting 100% success in test runs (with the exception of one case that had to be excluded, see #42061).

hwang37, 2022-01-24:
From our side, we can continue see this "fw downloading failure" issue. 
The "retry FW download" workaround is still needed. Can we keep this bug re-open?

We're using UP Extreme board as both Host Machine (on X86 CPU) and Tested Machine (Zephyr on Xtensa SOC).
[image]

AmberHibberd, 2022-01-24:
see above

enjiamai, 2022-01-25:
Verified on CAVS25, CAVS18 DSP, this issue is fixed by #41862 , and have also verified no impact on CAVS15.

closed 2022-01-25 by nashif