zephyrproject-rtos/zephyr #41864: ESP32 Wifi AP Mode DHCP Service

huarunlin, 2022-01-17:
I set the wifi working mode to ap mode, the mobile phone is connected to the esp32, and found that the mobile phone cannot obtain an IP through DHCP, how to start the DHCP service?

zephyr version: main branch c9a3dae6519fc788d33233fc1c4d1364ba129034

dkalowsk, 2022-01-18:
@huarunlin can you please provide for more details for resolving the issue?  Follow the steps at https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=bug&template=bug_report.md&title=

sylvioalves, 2022-01-19:
@huarunlin ESP32 WiFi SoftAP is not supported yet in Zephyr.

rftafas, 2022-01-19:
Going further: DHCP server is not supported also by Zephyr networking stack, not even for cabled (or the doc misses that, see link below). So, even if AP mode is supported, there will be the need to manually configure all IP addresses which seems messy. If the necessity for AP mode is to configure future IOT devices, I suggest:

-Using Bluetooth
-Use suitable host mode config (tethering, app,...)

PTAL: https://docs.zephyrproject.org/latest/guides/networking/overview.html#supported-features

Currently there is no short/mid plans to support DHCP server and AP mode.

huarunlin, 2022-01-20:
@rftafas Thanks,  looking forward to support in future updates

sylvioalves, 2022-03-18:
@huarunlin Closing this as we already have an issue opened regarding DHCP server: #38284

closed 2022-03-18 by sylvioalves

munir-zin, 2024-04-18:
Hi,
I need to get AP mode of ESP32 working. I am having the same issue of no ip address is assigned to client device when connecting to AP mode. Can anyone confirm if the support has been added for the DHCP or not? I am using `zephyr vv3.6.0`.

rftafas, 2024-04-18:
It has support since early this year (2024). All comments here are very outdated.

rftafas, 2024-04-18:
I suggest opening a new issue. This one is closed. Put all normal info for opening an issue there.