zephyrproject-rtos/zephyr #41874: Recursive spinlock error on ARM in specific circumstances

lairdjm, 2022-01-17:
**Describe the bug**
We are seeing an issue with zephyr from nRF connect SDK 1.8 which is based on zephyr 2.6 running on an nRF52840, we have code that is triggered from the shell which is used to restore a device to factory defaults and reboots, however upon doing this now (no issue was observed previously when using nRF connect SDK 1.6) after the warm reboot, it seems that a spinlock duplication error is emitted and the module reboots:
```
uart:~$ *** Booting Zephyr OS build v2.7.0-ncs1-6-ga0a5bd164a22  ***

uASSERTION FAIL [z_spin_lock_valid(l)] @ WEST_TOPDIR/zephyr/include/spinlock.h:129
        Recursive spinlock 0x20007328
[00:00:00.008,636] <inf> littlefs: LittleFS version 2.2, disk version 2.0
ASSERTION FAIL [z_spin_lock_valid(l)] @ WEST_TOPDIR/zephyr/include/spinlock.h:129
*** Booting Zephyr OS build v2.7.0-ncs1-6-ga0a5bd164a22  ***
```
According to the map file, that memory address is:
```
 .bss.sched_spinlock
                0x0000000020007328        0x4 zephyr/kernel/libkernel.a(sched.c.obj)
                0x0000000020007328                sched_spinlock
```
The way that this is triggered is shell command -> work submit -> function that restores to factory default by unlinking and writing files -> function that does reboot (LOG_PANIC, thread priority set to lowest level, sleep then warm reboot via sys_reboot()). If the fs_unlink function isn't then this spinlock error doesn't appear. So at this point I'm not sure what the cause is or why, if perhaps something isn't being initialised or if the error is in the application code.

A clear and concise description of what the bug is.

**To Reproduce**
Requires our application

**Expected behavior**
No spinlock issue to occur

**Impact**
Module recovers after a reboot which is automatic so doesn't cause any problems, but is just strange to see the issue

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: nRF connect SDK 1.8 which is using zephyr 2.7

lairdjm, 2022-01-17:
It seems that this is related to a watchdog thread causing a usage fault

lairdjm, 2022-01-17:
Something definitely amiss here, now have it reproducing with it originating in the QSPI driver, I don't think that it does originate where it says:
```
#0  k_spin_lock (l=0x20007318 <sched_spinlock>) at /tmp/AA/zephyr/include/spinlock.h:131
#1  z_unpend_first_thread (wait_q=wait_q@entry=0x20001988 <log_process_thread_sem>) at /tmp/AA/zephyr/kernel/sched.c:735
#2  0x0006c8b0 in z_impl_k_sem_give (sem=0x20001988 <log_process_thread_sem>) at /tmp/AA/zephyr/kernel/sem.c:93
#3  0x00044ed0 in esf_dump (esf=0x2001ae38 <z_interrupt_stacks+1976>) at /tmp/AA/zephyr/arch/arm/core/aarch32/fatal.c:26
#4  z_arm_fatal_error (reason=reason@entry=2, esf=esf@entry=0x2001ae38 <z_interrupt_stacks+1976>)
    at /tmp/AA/zephyr/arch/arm/core/aarch32/fatal.c:61
#5  0x00045e5a in z_arm_fault (msp=536981248, psp=<optimized out>, exc_return=4294967277, 
    callee_regs=0x2001aed0 <z_interrupt_stacks+2128>) at /tmp/AA/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1051
#6  0x000460c4 in z_arm_usage_fault () at /tmp/AA/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102
#7  <signal handler called>
#8  0x20000bb4 in qspi_nor_memory_data ()
#9  0x8deec600 in ?? ()
```

carlescufi, 2022-02-02:
Since I am not sure where this issue originates from, I have added multiple assignees here.
@lairdjm how can we try and reproduce this?

datho1801, 2022-03-27:
I encounter this issue many times recently. I did some debugging and see that:
+ The asserted object is `sched_spinlock `
+ The contained `l->thread_cpu` at the time being is `z_main_thread` from `init.c`
+ The issue will happen more frequently when having many interrupts (i2c interrupt...) during 'main` is executed
+ If the issue happens, it will almost appear right after 'main' has been done

So I suspect there is NO memory overlap here, the issue is `z_main_thread`'s scheduler invoked to lock `sched_spinlock` twice (or more):
```
if (thread_cpu != 0U) // thread_cpu = address of z_main_thread  { 
		if ((thread_cpu & 3U) == _current_cpu->id) // __current_cpu->id is always 0 { 
			return false;
		}
}
```

Environment:
+ OS Ubuntu 20.4
+ Zephyr 2.7

cvinayak, 2022-04-04:
I am seeing the issue on the latest upstream main 7b9ef30 by adding the below to `samples/bluetooth/peripheral_identity`  sample

```
CONFIG_BT_MAX_CONN=120
CONFIG_ASSERT=y
CONFIG_LOG=y
```

[image]
[image]

The different crashes are based on the value of `CONFIG_BT_MAX_CONN` above 102 (100 does not crash).

lairdjm, 2022-04-04:
@cvinayak I would say that's more likely to be because you're trying to allocate memory for 120 entries and it's overflowing

cvinayak, 2022-04-04:
> @cvinayak I would say that's more likely to be because you're trying to allocate memory for 120 entries and it's overflowing

@lairdjm No, all of them fit in the 256 KB RAM of nRF52840, otherwise build to generate the binary/hex should fail. `CONFIG_BT_MAX_CONN=120` for `samples/bluetooth/central_multilink` builds fine and successfully connects to peripherals.

cvinayak, 2022-04-04:
My issue is resolved, I was also using `CONFIG_BT_ID_MAX=120` which during `main()` execution made call to `bt_setup_random_id_addr()` consume over ~2.5KB of program stack in my main thread during the `bt_init()` . The overflow was too much corrupting other thread stacks and OS data memory. Increasing the `CONFIG_MAIN_STACK_SIZE` helped.

lairdjm, 2022-05-25:
Not seen issue in some time, closing issue and will reopen in future if it can be reproduced