zephyrproject-rtos/zephyr #41877: tests: kernel: fatal: ADSP stalls after firmware downloaded on Up Xtreme

mengxianglinx, 2022-01-17:
**Describe the bug**
Test cases in tests/kernel/fatal can stall the ADSP after firmware downloaded by cavs-fw-v25.py and can't be reset unless reboot. twister will be blocked at this point in daily testing.

 - Platform: ADSP on Up Xtreme

**To Reproduce**
Steps to reproduce the behavior:
1. cavsload.sh:
```
set -e

FLASHER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v25.py
LOGGER=$ZEPHYR_BASE/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
if [ "$1" = "" ]; then
        ssh root@localhost pkill -9 -f adsplog.py || true
        sleep 1
        ssh root@localhost "python3 -u $LOGGER --no-history || true"
        exit 0
fi

BLDDIR=$1

RIMAGE=$HOME/twister_tools/rimage
KEY=$HOME/twister_tools/keys/otc_private_key_3k.pem
IMAGE=$BLDDIR/zephyr/zephyr.ri
ELF=$BLDDIR/zephyr/main.mod
BOOT=$BLDDIR/zephyr/boot.mod

$RIMAGE -v -k $KEY -o $IMAGE -c $HOME/twister_tools/config/tgl.toml \
        -i 3 -e $BOOT $ELF > $BLDDIR/rimage.log

while [ ! -e $IMAGE ]; do
    sleep 0.1
done

ssh root@localhost pkill -9 -f cavs-fw-v25.py || true
sleep 1
ssh root@localhost "$FLASHER $IMAGE || true"
```
2. twister -p intel_adsp_cavs25 --device-testing --device-serial-pty=/path/to/cavsload.sh --west-flash=/path/to/cavsload.sh -T tests/kernel/fatal/
3. twister process will be blocked

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 44462723f5

andyross, 2022-01-18:
Is this a bug against the twister integration or the test case?  What happens when you run it manually?

andyross, 2022-01-18:
Took a look.  Indeed, exception handlers are having trouble on xtensa at the moment.  The proximate cause took a little bit of digging but is pretty obvious: the interrupt nesting count is zero in the level 1 interrupt (what other architectures call "exception") handler.  That causes the k_thread_abort() at the end of the fatal exception path to think it's in thread mode, and since SMP=y it enters a spin loop to synchronously wait for the termination of the thread it just aborted, which is the thread it interrupted (and thus can't run until the ISR is done).  And hijinks ensue.

I don't know why on TGL this cause a hard latchup of the DSP, that's just weird.  But it does.  On older APL hardware it's recoverable though so it's easier to debug there.

The root cause still eludes me.  I'm looking at the assembly right now and all interrupts share the same code path to increment the nest count.  Something else is going wonky; maybe something is scribbling on the CPU or there's a double-decrement happening somewhere.

mengxianglinx, 2022-01-19:
with the latest commit 5598ae85c0e3b29ed, tests/kernel/fatal dose not block twister any more, just failed.
But tests/kernel/tickless/tickless_concept blocks twister like tests/kernel/fatal before.

PR #41937 (2022-01-19): tests/kernel/fatal/exception: Remove legacy irq_lock() usage
The irq_lock() API is a legacy API not to be used for synchronization
by new code, and in any case is only being used in cargo-cult fashion
here.  These test cases all do synchronous exceptions, there's
literally nothing to synchronize against.

    (And in this case they're exposing a legacy wart.  On platforms where:

    1. SMP=y, which causes irq_lock() to be implemented as a somewhat
       complicated global lock

    2. No ARCH_EXCEPT() macro is defined, which causes the kernel to
       use a fallback that simply aborts the c [... cut]
commit 8d13be01 (2022-01-19): tests/kernel/fatal/exception: Remove legacy irq_lock() usage

The irq_lock() API is a legacy API not to be used for synchronization
by new code, and in any case is only being used in cargo-cult fashion
here.  These test cases all do synchronous exceptions, there's
literally nothing to synchronize against.

    (And in this case they're exposing a legacy wart.  On platforms where:

    1. SMP=y, which causes irq_lock() to be implemented as a somewhat
       complicated global lock

    2. No ARCH_EXCEPT() macro is defined, which causes the kernel to
       use a fallback that simply aborts the  [... cut]
closed 2022-01-19 by nashif