zephyrproject-rtos/zephyr #41879: new ztest api fails when user space is enabled

nordic-krch, 2022-01-17:
**Describe the bug**
It seems that new ztest API does not work with user space. I've modified one of test cases to use new ztest API if enabled. Test suite is shortened to only one test case. It fails when new api is enabled.

**To Reproduce**
Cherry-pick https://github.com/zephyrproject-rtos/zephyr/commit/67e5d72956a2dd8ed2f36e1a8d8020f4af65f0c9
1. Run it on `qemu_x86` and observe failure.
2. Comment `CONFIG_ZTEST_NEW_API=y` in `prj.conf` and run the test again. Test is passing.

**Expected behavior**
New ztest API should work with user space.

**Impact**
Transition to new API is blocked.

**Additional context**
New API introduced by #39714.

yperess, 2022-01-17:
https://github.com/zephyrproject-rtos/zephyr/commit/67e5d72956a2dd8ed2f36e1a8d8020f4af65f0c9#r64029060

I'm OK with making the before/after functions run in the same thread as the test. I think that would make sense.

yperess, 2022-02-08:
Resolved by #41985

closed 2022-02-08 by yperess