zephyrproject-rtos/zephyr #41899: ESP32 Wifi mDNS

huarunlin, 2022-01-18:
In my project, `CONFIG_MDNS_RESOLVER=y` is set, the wifi sta mode is connected to the wifi hotspot, and the ip is obtained. There is a reply to the host ping ip, but no reply to ping <hostname>.local. The host has installed mDNS related tools and can interpret other mDNS devices normally.
zephyr version: c9a3dae6519fc788d33233fc1c4d1364ba129034

dkalowsk, 2022-01-18:
@huarunlin thanks for the bug report.  It seems there are some details missing that you can hopefully provide such as build command.  The original template has a nice format to follow on it.

huarunlin, 2022-01-19:
> @huarunlin thanks for the bug report. It seems there are some details missing that you can hopefully provide such as build command. The original template has a nice format to follow on it.

**Describe the bug**
I made some modifications based on the samples\boards\esp32\wifi_station project, configured wifi ssid and password, enabled hostname, mDNS. esp32 successfully connected to wifi, and automatically assigned to ip. The host can access esp32 via ip, but fails to access via <hostname>.local

```
C:\Users\huarun.lin>ping RMS01-e89f6db9f080.local
Ping request could not find host RMS01-e89f6db9f080.local. Please check the name and try again.

C:\Users\huarun.lin>ping 192.168.1.140

Pinging 192.168.1.140 with 32 bytes of data:
Reply from 192.168.1.140: bytes=32 time=221ms TTL=64
Reply from 192.168.1.140: bytes=32 time=114ms TTL=64
Reply from 192.168.1.140: bytes=32 time=70ms TTL=64
Reply from 192.168.1.140: bytes=32 time=35ms TTL=64

Ping statistics for 192.168.1.140:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 35ms, Maximum = 221ms, Average = 110m
```

**To Reproduce**
1、modify prj.conf, configure the wifi settings and enable the hostname and mDNS functions.
```
CONFIG_WIFI=y
CONFIG_WIFI_ESP32=y
CONFIG_HEAP_MEM_POOL_SIZE=98304

CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_NET_LOG=y
CONFIG_NET_SHELL=n

CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_ECP_ALL_ENABLED=y

CONFIG_ESP32_WIFI_SSID="ssid"
CONFIG_ESP32_WIFI_PASSWORD="password"

CONFIG_NET_HOSTNAME="RMS01-"
CONFIG_NET_HOSTNAME_ENABLE=y
CONFIG_NET_HOSTNAME_UNIQUE=y
CONFIG_MDNS_RESOLVER=y
```

2、modify src\main.c, add hostname output 
```
#include <zephyr.h>
#include <sys/printk.h>
#include <esp_wifi.h>
#include <esp_timer.h>
#include <esp_event.h>

#include <net/net_if.h>
#include <net/net_core.h>
#include <net/net_context.h>
#include <net/net_mgmt.h>
#include <net/hostname.h>
[... 94 lines cut ...]
I (94) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=01870h (  6256) load
I (104) esp_image: segment 2: paddr=000118bc vaddr=3ffb1870 size=004ech (  1260) load
I (111) esp_image: segment 3: paddr=00011db0 vaddr=3ffb1d60 size=0006ch (   108) load
I (119) esp_image: segment 4: paddr=00011e24 vaddr=40080000 size=0e214h ( 57876) load
I (151) esp_image: segment 5: paddr=00020040 vaddr=3f400040 size=129e8h ( 76264) map
I (179) esp_image: segment 6: paddr=00032a30 vaddr=4008e214 size=00208h (   520) load
I (180) esp_image: segment 7: paddr=00032c40 vaddr=00000000 size=0d3d8h ( 54232) 
I (204) esp_image: segment 8: paddr=00040020 vaddr=400d0020 size=45400h (283648) map
I (314) boot: Loaded app from partition at offset 0x10000
I (314) boot: Disabling RNG early entropy source...
*** Booting Zephyr OS build v2.7.99-2893-gc9a3dae6519f  ***
[00:00:00.843,000] <inf> esp32_wifi_sta: Hostname RMS01-e89f6db9f080
[00:00:00.846,000] <inf> esp32_wifi: WIFI_EVENT_STA_START
[00:00:02.295,000] <inf> esp32_wifi: WIFI_EVENT_STA_CONNECTED
[00:00:05.561,000] <inf> net_dhcpv4: Received: 192.168.1.140
[00:00:05.561,000] <inf> esp32_wifi_sta: Your address: 192.168.1.140
[00:00:05.561,000] <inf> esp32_wifi_sta: Lease time: 86400 seconds
[00:00:05.561,000] <inf> esp32_wifi_sta: Subnet: 255.255.255.0
[00:00:05.561,000] <inf> esp32_wifi_sta: Router: 192.168.1.1
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Commit c9a3dae6519fc788d33233fc1c4d1364ba129034

closed 2022-04-10 by github-actions[bot]